raspberrypi/pico-examples #227: Building "Blink" in raspberry failed

JMR-Mejia, 2022-06-29:
I am following the steps of the raspberry pico tutorial(https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) but in the part of executing the following command:

```
make -j4
```
It throws me the following error and I can't find the error and I don't have much experience in c++
```
Scanning dependencies of target ELF2UF2Build
Scanning dependencies of target bs2_default
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
Scanning dependencies of target bs2_default_padded_checksummed_asm
[  0%] Generating bs2_default.bin
[  0%] No download step for 'ELF2UF2Build'
[  0%] Generating bs2_default_padded_checksummed.S
[  0%] No update step for 'ELF2UF2Build'
[  0%] Built target bs2_default_padded_checksummed_asm
[  0%] No patch step for 'ELF2UF2Build'
[  0%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - broken
CMake Error at /usr/share/cmake-3.18/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/gmake cmTC_b27a0/fast && gmake[3]: Entering directory '/home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp'
    /usr/bin/gmake  -f CMakeFiles/cmTC_b27a0.dir/build.make CMakeFiles/cmTC_b27a0.dir/build
    gmake[4]: Entering directory '/home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_b27a0.dir/testCCompiler.c.o
    /usr/bin/cc    -o CMakeFiles/cmTC_b27a0.dir/testCCompiler.c.o -c /home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp/testCCompiler.c
    as: error while loading shared libraries: /lib/aarch64-linux-gnu/libopcodes-2.35.2-system.so: invalid ELF header
    gmake[4]: *** [CMakeFiles/cmTC_b27a0.dir/build.make:85: CMakeFiles/cmTC_b27a0.dir/testCCompiler.c.o] Error 1
    gmake[4]: Leaving directory '/home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp'
    gmake[3]: *** [Makefile:140: cmTC_b27a0/fast] Error 2
    gmake[3]: Leaving directory '/home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp'

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeOutput.log".
See also "/home/main/SafeArea/raspberry/pico-examples/build/elf2uf2/CMakeFiles/CMakeError.log".
make[2]: *** [blink/CMakeFiles/ELF2UF2Build.dir/build.make:125: blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:4928: blink/CMakeFiles/ELF2UF2Build.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
```
I'm on part 3.1

aallan, 2022-06-29:
You either don't have a C compiler installed, or it can't find it.

JMR-Mejia, 2022-06-29:
> You either don't have a C compiler installed, or it can't find it.

What could I do?

JMR-Mejia, 2022-06-30:
> What platform and OS (and OS version) are you running on?

I am working on Raspberry Pi OS with desktop on a Raspberry Pi 4 model B

lurch, 2022-06-30:
`as: error while loading shared libraries: /lib/aarch64-linux-gnu/libopcodes-2.35.2-system.so: invalid ELF header`

Looks like something has gone with your installation somehow, so you might want to try again starting from a fresh SD card?

JMR-Mejia, 2022-07-01:
Reinstall the OS, but 64 bits and it worked for me

closed 2022-07-01 by JMR-Mejia