raspberrypi/pico-examples #231: Need a TLS example

peterharperuk, 2022-07-01:
I have a change that adds mbedtls support to the sd. We need an example to show how it's used.
See https://github.com/raspberrypi/pico-sdk/issues/893

dancesWithMachines, 2022-09-30:
Hello, I’d like to use the pending example https://github.com/peterharperuk/pico-examples/commit/450a4d51ae1dd5f571e06c636b1fda1ba80f6ace as library for my project (after modifications ofc). Do I need to wait until this will be merged with upstream? I am asking from legal point of view, as I need http request support asap.

aallan, 2022-09-30:
That code released, like most of the code released by us, is under the [BSD 3-clause](https://github.com/raspberrypi/pico-examples/blob/master/LICENSE.TXT) license. So long as you abide by the terms of this license you are free to use the code within these limits.

PR #305 (2023-02-04): Add tls_client example
This requires mbedtls to be added to the pico-sdk
Connects to worldtimeapi.org and retrieves a web page Originally written by Floris Bos @maxnet

Fixes: #231
closed 2023-05-17 by peterharperuk