raspberrypi/pico-examples #234: Cannot build the FreeRTOS examples

youca04, 2022-07-02:
The default behaviour for me was to not find FreeRTOS.h

D:/dev/rpi/pico-examples/pico_w/freertos/iperf/picow_freertos_iperf.c:14:10: fatal error: FreeRTOS.h: No such file or directory
   14 | #include "FreeRTOS.h"

If I add include paths to pico_w/freertos/iperf/CMakeLists.txt, then I get further 
+               ${FREERTOS_KERNEL_PATH}/include
+               ${FREERTOS_KERNEL_PATH}/portable/ThirdParty/GCC/RP2040/include

but cannot find a link library (
(xxx)/arm-none-eabi/bin/ld.exe: cannot find -lFreeRTOS-Kernel-Heap4
collect2.exe: error: ld returned 1 exit status

I'm obviously missing something - possibly config and/or build of FreeRTOS?

kilograham, 2022-07-02:
Passing `-DFREE_RTOS_KERNEL_PATH=xxx` to your `cmake` invocation is all that you should need.

Make sure you have the very latest `main` or `smp` branch of FreeRTOS-Kernel

youca04, 2022-07-03:
@kilograham Thank you very much for the fast response. I did have the kernel source cloned locally as part of the master FreeRTOS repo, but it was pointing to an older kernel (356fff8028734dc50a46972b2e315e2871b0a12e, V10.4.0-kernel-only-208-g356fff802), so I guess this tripped me up. Moving directly to the kernel repo allowed for a successful build.

Perhaps we should note to use the kernel repo directly, and not via FreeRTOS/FreeRTOS-SMP-Demos?

closed 2022-07-03 by youca04