raspberrypi/pico-examples #235: There is no data in hello_usb example

Moon-Haze, 2022-07-07:
I test the pico 's uart/usb. I  write code  according to the example. But I don't receive data using serial's software.
I use WSl to build the project. Does the problem  occur in WSL? I also don't receive data using the hello_usb example.
```
# CMakeLists.txt
cmake_minimum_required(VERSION 3.15)
include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)
project(pico_camke_test C CXX ASM)

set(CMAKE_C_STANDARD 11)

pico_sdk_init()

add_executable(${PROJECT_NAME} main.c)

pico_add_extra_outputs(${PROJECT_NAME})

pico_enable_stdio_usb(${PROJECT_NAME} 1)
pico_enable_stdio_uart(${PROJECT_NAME} 0)

target_link_libraries(${PROJECT_NAME} pico_stdlib)
```
```
// main.c
#include <stdio.h>

#include "pico/stdlib.h"

int main() {
#ifndef PICO_DEFAULT_LED_PIN
#warning blink example requires a board with a regular LED
#else
    const uint LED_PIN = PICO_DEFAULT_LED_PIN;
    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);
    stdio_init_all();
    while (true) {
        gpio_put(LED_PIN, 1);
        printf("LED ON \r\n");
        sleep_ms(250);
        gpio_put(LED_PIN, 0);
        printf("LED OFF \r\n");
        sleep_ms(250);
    }

#endif
}
```

lurch, 2022-07-07:
Are you sure you've connected to the right COM port? See https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf for detailed instructions.

lurch, 2022-07-07:
Please carefully follow the instructions in the getting-started-guide (i.e. build _without_ using WSL) and let us know if it's still not working?
When you say "I also don't receive data using the hello_usb example" was that a version of hello_usb you built yourself using WSL, or using the https://datasheets.raspberrypi.com/soft/hello_world.uf2 binary?

Moon-Haze, 2022-07-07:
> Please carefully follow the instructions in the getting-started-guide (i.e. build _without_ using WSL) and let us know if it's still not working?
> When you say "I also don't receive data using the [...]

Thank you. I use my built using the WSL.I will test the release binary file to test.