raspberrypi/pico-examples #245: Pico W Can't Send/Receive More Than 1460 Bytes

mucahitkurtlar, 2022-07-21:
I tried `pico_w/tcp_server/` and `pico_w/tcp_client/` examples (flashed picow_tcpip_*_background.uf2 ones) with my own python test scripts (because `pico_w/python_test_tcp/` written in MicroPython) but board couldn't send/receive default BUF_SIZE=2048. Then I tried with the 1024 bytes everything is okay but I don't understand why the limit is 1460 bytes. What am I doing wrong?

lurch, 2022-07-21:
ping @peterharperuk

peterharperuk, 2022-07-21:
Can you attach your script? It sounds like an issue with the tcp maximum segment size. It's possible the examples need to be improved.

mucahitkurtlar, 2022-07-21:
> Can you attach your script? It sounds like an issue with the tcp maximum segment size. It's possible the examples need to be improved.

I've already created a pull request here https://github.com/raspberrypi/pico-examples/pull/246

Also this example uses 2048 bytes. If examples will be changed I can update my pr.

peterharperuk, 2022-07-25:
I have fixed your scripts here https://github.com/raspberrypi/pico-examples/compare/master...peterharperuk:pico-examples:add-python-tests

The MTU is 1460, so you'll only get a buffer of this size. I have some linux examples somewhere that behave the same.

lurch, 2022-07-26:
@peterharperuk Just to clarify... are you going to be PRing your branch, or are you expecting @mucahitkurtlar to pull your fixes into his existing PR #246 ?

peterharperuk, 2022-07-26:
I'm easy. Happy for @mucahitkurtlar to update his PR. I fixed a few things in my change.

mucahitkurtlar, 2022-07-26:
Should I update my PR myself or pull changes from @peterharperuk. If I try to pull from Peter I have to resolve conflict.

mucahitkurtlar, 2022-07-26:
Ok, I resolved the conflict and update my PR. @lurch @peterharperuk

Yang123bing, 2022-08-02:
there are two pio modules and 8 state machine in the rp2040,and where I can find the state machine's source code? I wanna know how the state machine works.

lurch, 2022-08-02:
(For the benefit of other people in this thread, @Yang123bing 's question has been answered here: https://github.com/raspberrypi/pico-feedback/issues/267 )

closed 2023-02-27 by mucahitkurtlar