raspberrypi/pico-examples #247: Compilation errors in cyw43_lwip.c

holographicterminal, 2022-07-23:
When making the pico_w/tcp_server example, cyw43_lwip.c has several compilation errors.

CMake seems to work and this is the output after running make:

`/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c:158:5: warning: implicit declaration of function 'netif_set_hostname'; did you mean 'netif_set_down'? [-Wimplicit-function-declaration]
  158 |     netif_set_hostname(n, CYW43_HOST_NAME);
      |     ^~~~~~~~~~~~~~~~~~
      |     netif_set_down
/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c:180:5: warning: implicit declaration of function 'cyw43_hal_get_mac_ascii'; did you mean 'cyw43_hal_get_mac'? [-Wimplicit-function-declaration]
  180 |     cyw43_hal_get_mac_ascii(CYW43_HAL_MAC_WLAN0, 8, 4, &mdns_hostname[4]);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
      |     cyw43_hal_get_mac
/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c:182:5: error: too many arguments to function 'mdns_resp_add_netif'
  182 |     mdns_resp_add_netif(n, mdns_hostname, 60);
      |     ^~~~~~~~~~~~~~~~~~~
In file included from /.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c:42:
/home/.../pico-sdk/lib/lwip/src/include/lwip/apps/mdns.h:110:7: note: declared here
  110 | err_t mdns_resp_add_netif(struct netif *netif, const char *hostname);
      |       ^~~~~~~~~~~~~~~~~~~
/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c: In function 'cyw43_cb_tcpip_deinit':
/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c:201:28: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
  201 |             netif->ip_addr.addr = 0;
      |                            ^~~~
      |                            u_addr
/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c: In function 'cyw43_tcpip_link_status':
/.../pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c:239:28: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
  239 |         if (netif->ip_addr.addr != 0) {
      |                            ^~~~
      |                            u_addr
`

I've spent awhile looking into this problem myself but haven't found a solution yet.

It seems like a problem with the source code but it might be possible that I am not including something properly.

This shouldn't have anything to do with how it's made with CMake, but I would also appreciate it if someone had a guide on how to make this example so that I can be sure that this isn't a problem caused by the improperly making it.

lurch, 2022-07-23:
I've just tried this myself (on my Ubuntu 18.04 laptop) with a fresh checkout of the `master` branches of `pico-sdk` and `pico-examples`, and it all built fine. Is it possible that some of your submodules are out of date?

For the avoidance of any doubt, I did:
```
cd /tmp
mkdir pico
cd pico
git clone https://github.com/raspberrypi/pico-sdk
git clone https://github.com/raspberrypi/pico-examples
cd pico-sdk
git submodule update --init
cd ../pico-examples
mkdir build
cd build
PICO_SDK_PATH=../../pico-sdk cmake -DPICO_BOARD=pico_w -DWIFI_SSID=my-ssid -DWIFI_PASSWORD=my-password ..
cd pico_w/tcp_server
make
```

holographicterminal, 2022-08-05:
Sorry for the delay.

Yes! This worked to fix the problem.  Thanks!

closed 2022-08-05 by holographicterminal