raspberrypi/pico-examples #248: pico-spi sending

qombajn, 2022-07-28:
Hello!
I want to run PGA2311U with Raspberry Pi Pico and I don't know how to setup spi:

uint8_t r=50;
uint8_t l=150;
spi_init(spi0,6250000); //6.25Mhz
gpio_set_function(PIN_MISO, GPIO_FUNC_SPI);
gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
gpio_set_function(PGA_CS, GPIO_FUNC_SPI);
spi_set_format(spi0,8,SPI_CPOL_0,SPI_CPHA_0,SPI_MSB_FIRST);
printf("spi try send  %d %d\n",r,l);
gpio_put(PGA_CS,0);
spi_write_blocking(spi0,&r,1);
spi_write_blocking(spi0,&l,1);
gpio_put(PGA_CS,1);

I don't know what is wrong in here. :( 
Here is PGA2311U Datasheet:
 [https://www.ti.com/lit/ds/symlink/pga2311.pdf?ts=1658994595000&ref_url=https%253A%252F%252Fwww.google.com%252F](url)
[image]
[image]
[image]

lurch, 2022-07-28:
This is probably the sort of question better suited for https://forums.raspberrypi.com/viewforum.php?f=143

qombajn, 2022-07-31:
Problem was bad wiring GND.

closed 2022-07-31 by qombajn