raspberrypi/pico-examples #251: build error for pico_w and freertos

efzedvi, 2022-08-03:
I am getting a build error when I try to make this repo for FreeRTOS and Pico W.

Here is what I am doing, I run under the build directory:  
cmake -DPICO_BOARD=pico_w -DWIFI_SSID=abc -DWIFI_PASSWORD=xyz  ..
(I make sure FREERTOS_KERNEL_PATH points to where FreeRTOS source is)
and then I get this when I run make:

`[ 46%] Building C object pico_w/freertos/iperf/CMakeFiles/picow_freertos_iperf_server_sys.dir/picow_freertos_iperf.c.obj
/home/efzedvi/local/src/pico/pico-examples/pico_w/freertos/iperf/picow_freertos_iperf.c:14:10: fatal error: FreeRTOS.h: No such file or directory
   14 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [pico_w/freertos/iperf/CMakeFiles/picow_freertos_iperf_server_sys.dir/build.make:63: pico_w/freertos/iperf/CMakeFiles/picow_freertos_iperf_server_sys.dir/picow_freertos_iperf.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:6988: pico_w/freertos/iperf/CMakeFiles/picow_freertos_iperf_server_sys.dir/all] Error 2
make: *** [Makefile:84: all] Error 2`

Could anyone please tell me what I am missing or doing wrong?

Thanks

lurch, 2022-08-31:
Duplicate of #234 ?

peterharperuk, 2022-11-10:
Despite what I say above, I usually define it via an environment variable. Are you using the full absolute path?

AndrewCapon, 2022-11-10:
Yep full absolute path, this is on a Mac with big sur.