raspberrypi/pico-examples #252: USB CDC Host error

freespirit83xyz, 2022-08-05:
Hello,
I'm new in raspberry dev. I need to develop a communication with an usb device. I builded Host_cdc_msc_hid example, it runs but if I plug in a mouse or keyboard nothing happens. If I plug in my device raspberry crashes.
I attach debugger log:

target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
Unable to parse SVD file /src/rp2040/hardware_regs/rp2040.svd: Error: ENOENT: no such file or directory, open '/src/rp2040/hardware_regs/rp2040.svd'
Note: automatically using hardware breakpoints for read-only addresses.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000138 msp: 0x20041f00

Thread 
1 received signal SIGTRAP, Trace/breakpoint trap.
_exit (status=status@entry=1) at /Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:183
183	        __breakpoint();

And this is the stack of calls

_exit@0x10002004 (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:183)
__assert_func@0x1000206e (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:228)
hw_trans_complete@0x10004b6c (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c:164)
hcd_rp2040_irq@0x10004c02 (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c:212)
<signal handler called>@0xfffffff9 (Origine sconosciuta:0)
divmod_u32u32_unsafe@0x10003736 (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/rp2_common/pico_divider/divider.S:186)
divmod_u32u32_savestate@0x1000376a (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/rp2_common/pico_divider/divider.S:217)
us_to_ms@0x100005fa (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/common/pico_time/include/pico/time.h:71)
to_ms_since_boot@0x100005fa (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/src/common/pico_time/include/pico/time.h:84)
board_millis@0x100005fa (/Users/motusbookpro/Documents/Lavori/pico/pico-sdk/lib/tinyusb/hw/bsp/board.h:91)
led_blinking_task@0x100005fa (/Users/motusbookpro/Documents/Lavori/pico/pico-examples/usb/host/host_cdc_msc_hid/main.c:123)
main@0x10000644 (/Users/motusbookpro/Documents/Lavori/pico/pico-examples/usb/host/host_cdc_msc_hid/main.c:54)

What can I do?

Thanks in advance

Tails86, 2023-01-26:
I'm seemingly running into the same thing. As soon as I plug in a mouse/keyboard, the following shows on the serial output.

`assertion "ep->active" failed: file "/home/user/github/usb_host_test/ext/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 164, function: hw_trans_complete`

Tails86, 2023-01-26:
I noticed that this file was updated on master of tinyusb: https://github.com/hathach/tinyusb/blob/master/src/portable/raspberrypi/rp2040/hcd_rp2040.c#L170

Moving the assert inside the `if` context also gets this to work.

Tails86, 2023-02-05:
Workaround: disable assert() by defining NDEBUG on tinyusb_host 

`target_compile_definitions(tinyusb_host INTERFACE NDEBUG=1)`

Tails86, 2023-02-16:
I am no longer seeing this issue with version 1.5 of the pico-sdk and associated pico-examples

closed 2023-02-16 by kilograham