raspberrypi/pico-examples #260: pico w examples do not get built

tiborh, 2022-08-26:
good up to this point:
1. create build directory (mkdir build)
2. cd build
3. cmake ..
4. then other make examples seem to be ok (e.g. hello_world or blink)
5. however make in picow directory does not build anything
any idea? what logs would be needed to make sure what is wrong?

kilograham, 2022-08-27:
`cmake -DBOARD=pico_w ..` 

It does not actually log anything, perhaps it could (just seemed like spam for people who weren't using pico_w)

tiborh, 2022-08-27:
great, thanks.
two new lines appeared:
```
Enabling build support for Pico W wireless.
Skipping some Pico W examples as WIFI_SSID is not defined
```

kilograham, 2022-08-27:
https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf

tiborh, 2022-08-28:
Thank you. This is also great. A good second after the starter guide.

efzedvi, 2022-08-31:
The freeRTOS examples however do not get built, as stated here:
https://github.com/raspberrypi/pico-examples/issues/251

lurch, 2022-08-31:
@tiborh Can this be closed now?

tiborh, 2022-08-31:
oh. yes. sorry. issue solved.

closed 2022-08-31 by tiborh