raspberrypi/pico-examples #262: VSCode launch.json templates "runToMain" deprecated

jamd315, 2022-08-31:
In the templates launch-raspberrypi-swd.json and launch-remote-openocd.json under /ide/vscode, both use the deprecated debug attribute `runToMain: true`, which causes a warning to display in VSCode.  See also [cortex-debug debug attributes](https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md)

[image]

lurch, 2022-08-31:
ping @JamesH65

JamesH65, 2022-09-01:
Yes, definitely a valid issue. Not sure if I can add the new command, which would be backwards compatible, or need to replace the runtomain completely, which is no so backwards compatible.