raspberrypi/pico-examples #266: add freertos httpd example

peterharperuk, 2022-09-02:
I added this for some reason - may as well have another free rtos example?

cmarcante, 2022-09-22:
Hello,
The examples on Internet for **HTTPD** has problems or errors when you try to compile them.
Please, add an example made from  Raspberry.org
Thanks.
-------------------------------------------------------------------------------------------------------------
This is an example of error without solution when you compile the demos software from internet.
To solve this problem, you need to know too much the lwIP to solve this problem.
 
assignment to 'struct fsdata_file *' from incompatible pointer type 'struct fs_file *' [-Werror=incompatible-pointer-types]

struct fsdata_file {
  const struct fsdata_file *next;
  const unsigned char *name;
  const unsigned char *data;
  int len;
  u8_t flags;
#if HTTPD_PRECALCULATED_CHECKSUM
  u16_t chksum_count;
  const struct fsdata_chksum *chksum;
#endif /* HTTPD_PRECALCULATED_CHECKSUM */
};

struct fs_file {
  const char *data;
  int len;
  int index;
  /* pextension is free for implementations to hold private (extensional)
     arbitrary data, e.g. holding some file state or file system handle */
  fs_file_extension *pextension;
#if HTTPD_PRECALCULATED_CHECKSUM
  const struct fsdata_chksum *chksum;
  u16_t chksum_count;
#endif /* HTTPD_PRECALCULATED_CHECKSUM */
  u8_t flags;
#if LWIP_HTTPD_CUSTOM_FILES
  u8_t is_custom_file;
#endif /* LWIP_HTTPD_CUSTOM_FILES */
#if LWIP_HTTPD_FILE_STATE
  void *state;
#endif /* LWIP_HTTPD_FILE_STATE */
};

peterharperuk, 2022-09-22:
>> Please, add an example made from Raspberry.org
I don't understand this - sorry. It compiled fine last time I tried it. Maybe you can include the steps that led up to your problem?

kilograham, 2024-07-22:
merged into devleop

PR #458 (2024-07-22): Add lwip httpd examples
Fixes #266
Fixes #272
commit 1b895e1b (2024-07-22): Add lwip httpd examples (#458)

Fixes #266
Fixes #272
closed 2024-07-22 by kilograham