raspberrypi/pico-examples #272: Add a test for mDns

peterharperuk, 2022-09-08:
I don't fully understand mDns but I think it gives me the ability to register a local domain name, so a web server can register and you can surf to it like http://picow1a28.local/?

Anyway - there's code in the SDK wrapped in #if LWIP_MDNS_RESPONDER, so we should test it works?

lurch, 2022-09-08:
It seems like that's the `#if` that will cause https://github.com/georgerobotics/cyw43-driver/issues/17 to fall over?

lurch, 2022-12-09:
I guess this should be closed, now that mDNS has been removed by https://github.com/georgerobotics/cyw43-driver/pull/39 ?

peterharperuk, 2022-12-09:
It was removed from the driver, but I was going to add it to one of the examples.

lurch, 2022-12-09:
Ah, apologies for my misunderstanding. So this will get fixed by #269 ?

peterharperuk, 2022-12-09:
Yes - let me check if it works now.

kilograham, 2024-07-22:
merged into develop

kilograham, 2024-07-22:
actually - @peterharperuk is this fixed?

peterharperuk, 2024-07-22:
I'm going to say yes (I just tested it). The httpd example you just pushed uses it. It prints out the host name...
mdns host name PicoW7282.local
If you are on the same subnet opening PicoW7282.local in a web browser takes you to the website. That's my understanding of mDNS.

PR #458 (2024-07-22): Add lwip httpd examples
Fixes #266
Fixes #272
commit 1b895e1b (2024-07-22): Add lwip httpd examples (#458)

Fixes #266
Fixes #272
closed 2024-07-22 by peterharperuk