raspberrypi/pico-examples #278: undefined reference to `cyw43_tcpip_link_status' when trying to build ntp-client with Arduino IDE

lu7did, 2022-09-18:
When trying to compile the program ntp-client with the Arduino IDE and after few adaptations to allow for that (comment stdio_init_all() and a couple of castings as well as introduce setup() and loop()) the compilation fails at linking time with the following message

...test\ntp_client/cyw43_arch.c:79: undefined reference to `cyw43_tcpip_link_status'

The prototype for that function is at cyw43.h but including it won't solve the problem
The function itself is at cyw43_lwip.c but it should be in the library path, including it explicitly at the program folder won't solve the problem and actually starts to generate an explosion of dependency issues.

Having search for a similar problem elsewhere but to no avail. It seems a similar issue might prevent the usage of any example on the pico_w path.

I've been working successfully with the IDE, including adaptations of several other examples on the pico-examples path (specially dma, multicore and PIO).

¿Any help or suggestion?

Thanks and best regards, Pedro.

peterharperuk, 2022-09-18:
On the command line at least, you might see that if you try building something that uses wi-fi but the SDK thinks you're building for a plain Pico. Have you defined PICO_BOARD=pico_w ?

lu7did, 2022-09-18:
Yes, I did.
[image]

lu7did, 2022-09-18:
Yes, I did
[image]

aallan, 2022-09-18:
Which version of the Arduino IDE are you using? The official one from Arduino, which sits on top of Arm Mbed, or the community port which sits directly on top of the Pico C SDK?

lu7did, 2022-09-18:
Thank you for your answer. The Arduino version I'm using is 1.8.19. The libraries used are 

[image]

I understand these are what you refer as the community port and it's using the Pico C SDK. In fact I has been using other parts of the pico-examples without issues (dma, core, pio, serial) it's just with pico_w what is creating troubles.

The code I'm developing needs to be made in C/C++ as it´s just a part to be later integrated in a larger project.

What is the forum you suggest to be the most appropiate for my query?

Thanks, Pedro.

acidclear, 2023-02-14:
@lu7did if this is still an issue for you, i solved it by replacing `cyw43_arch_none` with `pico_cyw43_arch_lwip_threadsafe_background` in my CMakeLists file