raspberrypi/pico-examples #281: Pico PIO - multiple encoders

MadTooler, 2022-10-04:
I am still trying to wrap my head around the language of PIO code for the Pico.  I have the example for the [quadrature encoders](https://github.com/raspberrypi/pico-examples/tree/master/pio/quadrature_encoder) working for a single encoder, but I am unclear the method I should use to declare a second or more.

I am using PlatformIO within VScode for arduino style C++ format.  I found https://wokwi.com/tools/pioasm to be easiest in compiling the PIO code to a .h header file to use with PlatormIO.

Beyond adding extra encoders, I would also like to make a modified version that allows me to read single edge encoders for my purpose of reading step/dir outputs from a motion controller so I can make a custom closed loop drive.  I do not know if there is enough room in the PIO program to fit the needed optional lines of instructions or if I will need to make a dedicated version.

paulwratt, 2022-10-04:
[Raspberry Pi Pico PIO - 8 Little Processors You Can Program](https://www.youtube.com/watch?v=QlKtEA5XKc4)

MadTooler, 2022-10-04:
Thanks.  I have been educating myself how to work with the PIOs.

I am not clear on if the quadrature encoder example can be called to run additional encoders, or if it is essentially hardcoded to only be able to run a single.  As in, do I need to make separate PIO programs hardcoded to different state machines for each encoder?

closed 2022-10-04 by kilograham