raspberrypi/pico-examples #286: Hello Usb example not working ubuntu 22

geoeo, 2022-10-23:
I can not get "hello world" response from screen or minicom in Ubuntu 22.

I can confirm it works because I can get a response via Hercules in Windows 11.
So somehow the steps in the published guide are wrong or incomplete.

daniilsunyaev, 2023-03-29:
Had similar issue, and noticed that `minicom -b 115200 -o -D /dev/ttyACM0` works if you run it as sudo.
That is probably related to user restrictions. In my case `stat /dev/ttyACM0` shows that it is attached to `dialout` group: `Access: ... Gid: (   20/ dialout)`.
Users are not in `dialout` group by default. When I've added my user to this group, minicom command started working as expected.

lurch, 2025-04-26:
Closing as user-problem with permissions, not bug in pico-examples

closed 2025-04-26 by lurch