raspberrypi/pico-examples #295: Creating a folder for my projects gives me CMake Error

seamusdemora, 2022-12-03:
Rather than cluttering my `pico-examples` folder with my efforts to create my own project, I'm trying to follow instructions from Chap 7 (Creating your own Project) of the `Getting started with Raspberry Pi Pico` document.

I've done it a bit differently in that I've created the folder `~pico/MyProjects`, and then added a subfolder `~pico/MyProjects/ina260_i2c` to house my first project. In other words, the `MyProjects` folder is at the same level under `~/pico` as the `pico-examples` folder is. IAW the instructions, I've also added a `~pico/MyProjects/ina260_i2c/build` subfolder. 

Further IAW the instructions, I've added my source file (`ina260_trial.c`), and the following `CMakeLists.txt`:

```
add_executable(ina260_trial
        ina260_trial.c
        )

# pull in common dependencies and additional i2c hardware support
target_link_libraries(ina260_trial pico_stdlib hardware_i2c)

# this enables stdio via minicom on my Ubuntu Linux box
pico_enable_stdio_usb(ina260_trial 1)	    # Why does this throw an error??
# pico_enable_stdio_usb(i2c_bus_scan 1)  # from the bus_scan Example

# create map/bin/hex file etc.
pico_add_extra_outputs(ina260_trial)

# add url via pico_set_program_url
example_auto_set_url(ina260_trial)
```

This CMakeLists.txt file was copied from one of the `i2c_bus_scan` example in `pico-examples`; adjusted for my ina260_trial project, and using the usb output instead of the serial port (this allows me to use `minicom` on by development system (Ubuntu). However, I get warnings and an error when `cmake is run. It seems that I am missing something, but except for the added folder depth, I've followed the instructions in the `Getting Started...` document. 

What am I missing - where did I go astray & how to fix this?

lurch, 2022-12-03:
I assume you mean Chapter 8 of the Getting Started guide? :wink: 
You didn't mention whether you'd copied across the `pico_sdk_import.cmake` ?

This is really the sort of question you ought to be asking at https://forums.raspberrypi.com/viewforum.php?f=143 rather than on GitHub issues.

seamusdemora, 2022-12-04:
@lurch ,

>I assume you mean Chapter 8 of the Getting Started guide? 😉
Umm, no - Chapter 7 is "Chapter 7. Creating your own Project", and that's what I was trying to do.  I thought putting my projects in a different folder from pico-examples would be best - not to get balled up w/ git updates, PRs, etc. 

>You didn't mention whether you'd copied across the pico_sdk_import.cmake ?
Yes, I did. Apologies for omitting that from my question. I've edited the question - hope that's clear now. 

>This is really the sort of question you ought to be asking at https://forums.raspberrypi.com/viewforum.php?f=143 rather than on GitHub issues. 
Probably - but there's a reason I don't, and a long history to go with it. As briefly as possible, I am "double-banned for life" in the forums. First time due to a difference of opinion I had with someone, second time was after Phil Elwell told me that he checked & I was not banned. So I returned to the forum for a couple of days when the "gatekeepers" discovered me, got really upset & accused me of fraud, and then told me, "No chance of parole". So, maybe I should be banned for life here also? I hope not, but whatever Lord Raspberry decides, I'll live with it.

lurch, 2022-12-04:
> Umm, no - Chapter 7 is "Chapter 7. Creating your own Project"

Are you looking at an old version of the [Getting Started PDF](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)? :thinking: 
[image]

> This CMakeLists.txt file was copied from one of the i2c_bus_scan example in pico-examples

I've just re-read your original question, and that won't work, as the `CMakeLists.txt` used in `pico-examples` is slightly different to the `CMakeLists.txt` used by a standalone project. I suggest you re-read Chapter 8 in more detail, and start from the `CMakeLists.txt` given there, rather than trying to copy one from pico-examples.