raspberrypi/pico-examples #299: examples failing to build in debug

peterharperuk, 2022-12-06:


peterharperuk, 2022-12-06:
> /home/peterh/bin/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/ht16k33_i2c.dir/ht16k33_i2c.c.obj: in function `main':
> 
> /home/peterh/source/pico/pico-examples3/i2c/ht16k33_i2c/ht16k33_i2c.c:166: undefined reference to `ht16k33_display_set'
> 
> /home/peterh/bin/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/peterh/source/pico/pico-examples3/i2c/ht16k33_i2c/ht16k33_i2c.c:167: undefined reference to `ht16k33_display_set'
> 
> /home/peterh/bin/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/peterh/source/pico/pico-examples3/i2c/ht16k33_i2c/ht16k33_i2c.c:168: undefined reference to `ht16k33_display_set'
> 
> /home/peterh/bin/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/peterh/source/pico/pico-examples3/i2c/ht16k33_i2c/ht16k33_i2c.c:169: undefined reference to `ht16k33_display_set'
> 
> collect2: error: ld returned 1 exit status
> 
> make[2]: *** [i2c/ht16k33_i2c/CMakeFiles/ht16k33_i2c.dir/build.make:823: i2c/ht16k33_i2c/ht16k33_i2c.elf] Error 1
> 
> make[1]: *** [CMakeFiles/Makefile2:5738: i2c/ht16k33_i2c/CMakeFiles/ht16k33_i2c.dir/all] Error 2
> 
> make: *** [Makefile:91: all] Error 2
> 
>

lurch, 2022-12-06:
I guess this can be closed if the fix has been merged?

PR #300 (2022-12-06): Fix debug build issue
Fixes #299
commit b1dae7fb (2022-12-06): Fix debug build issue (#300)

Fixes #299
closed 2022-12-06 by JamesH65