raspberrypi/pico-examples #318: Feature request: http and https client examples in C

a-h, 2023-01-22:
I want to post data from sensors over HTTPS in JSON format, and ideally, also using MQTT.

This seems very difficult with the C API, but is trivial with the `urequests` API in MicroPython.

On the ESP32, the included `WiFiClientSecure` type makes the same task simple too.

So far, I've spent hours trying to get something up and running on the Pico W without success.

Reading through https://www.i-programmer.info/programming/hardware/15838-the-picow-in-c-simple-web-client.html got me up and running with a HTTP request, but that's not very useful, since a lot of cloud infrastructure doesn't support unencrypted transports. Regardless, I think it would be a good addition to the examples.

I've looked at https://github.com/cniles/picow-iot but couldn't get it to build. It seems outdated, since the Pico SDK seems to have some support for mbedTLS now. The author didn't get any help on his questions about performance at https://forums.raspberrypi.com/viewtopic.php?t=339620 so I'm not sure if it works anyway.

This unmerged PR seems like it's starting to get to the right place, but it's still very much at the TCP side, rather than higher level. https://github.com/raspberrypi/pico-examples/pull/305 - it doesn't make use of the httpc_get_file_dns function, for example.

Reading through this blog post makes it look like it's possible to do HTTPS, but I don't know how to connect the pieces -
https://www.unshiu.com/posts/pico-http-client-part-iii-mbedtls/ - Finally, it seems that the author gave up on the Pico SDK, and switched to try out FreeRTOS (https://www.unshiu.com/posts/pico-with-freertos/)

I think there's two things that would help the developer experience here:

1. Examples of HTTP and HTTPS connectivity with the existing tools.
2. Addition of a higher level API for HTTPS requests built into the Raspberry Pi SDK to simplify it.

a-h, 2023-01-23:
`arduino-pico` has a good HTTP API that's borrowed from the ESP8266, but all the Arduino libraries are LGPL licensed.

* https://arduino-pico.readthedocs.io/en/latest/httpclient.html

```C
// Error checking is left as an exercise for the reader...
HTTPClient https;
https.setInsecure();  // Use certs, but do not check their authenticity
if (https.begin("https://my.secure.server/url")) {
    if (https.GET() > 0) {
        String data = https.getString();
    }
    https.end();
}
```

NworbLegin, 2023-03-19:
I too would like some examples specificall for the Pico in C with HTTP and HTTPS GET/POST requestes for this as I have attempted to use the httpc_get_file function from #include "lwip/apps/http_client.h", but this seems to fall over after a number of repeated requests with memory issues.

pspeybro, 2023-04-06:
See my comment [here](https://stackoverflow.com/questions/26192758/how-can-i-send-a-simple-http-request-with-a-lwip-stack/75264534#75264534)
Based on [this](https://www.i-programmer.info/programming/hardware/15838-the-picow-in-c-simple-web-client.html) example, which also has the memory leak.

You will need to free the memory taken by the struct pbuf *hdr in the headers function and struct pbuf *p in the body function with respectively pbuf_free(hdr); and pbuf_free(p);

marceloalcocer, 2023-05-29:
I wrote a minimum working [HTTPS client example in C](https://github.com/marceloalcocer/picohttps) which might be useful as a starting point for an example.

It's currently in it's own repo, but would be happy to migrate it here if desired. Might require some refactoring though.

Looks like @peterharperuk is also working on adding some [HTTP and HTTPS examples](https://github.com/peterharperuk/pico-examples/tree/add_http_example) though which might be ready to merge soon.

fwhigh, 2023-12-30:
Any chance @peterharperuk 's examples can get completed and merged soon?

lavara123, 2025-08-10:
What's the latest?

 I need to post to Firestore (Cloud Function) with my Pi Pico 2w.

lavara123, 2025-08-11:
> http examples were merged. But the LwIP http client doesn't support post so that doesn't help you much.

So how can I do this?
Post -> firestore cloud functions

peterharperuk, 2025-08-11:
I would probably modify the LwIP http client to do what I want. It might be quicker to look for another http client that meets your needs.

lavara123, 2025-08-11:
> I would probably modify the LwIP http client to do what I want. It might be quicker to look for another http client that meets your needs.

When you say another solution, do you mean using an additional ESP32?

peterharperuk, 2025-08-11:
No. I mean look in github for a better http client

lavara123, 2025-08-11:
> No. I mean look in github for a better http client

I am new to this, what other https clients are available for raspberry pi pico 2w

pspeybro, 2025-08-11:
I managed to get HTTP POST working based on some code I found elsewhere (can't immediately locate the source).
The code below will probably not compile, but I stripped my program to the basics to show how I have it right now. 
I use it to post data to influxdb. 
No HTTPS in this example though.

CMakeLists.txt references the pico_lwip_http library.

I hope it can help someone.

`#include <stdio.h>
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"
#include "lwip/apps/http_client.h"  //http client
#include "lwip/altcp_tls.h"
#include "lwip/tcpip.h"
#include "lwip/init.h"
#include "pico/cyw43_arch/arch_threadsafe_background.h"  

#define POST_URL "192.168.1.1"
#define POST_PATH "/api/v2/write?bucket=Greenhouse&org=MyOrg&precision=s"
#define BEARER_TOKEN "influx_api_token_here=="

#define PORT 49999

struct tcp_pcb *pcb;
ip4_addr_t server_ip;

float Temperature1;
float Temperature2;
  

const char ssid[] = "";
const char pass[] = "";
const uint32_t auth = CYW43_AUTH_WPA2_MIXED_PSK;
const uint32_t country = CYW43_COUNTRY_BELGIUM;

bool httpBusy = false;
char httpBuff[2000]; //if not big enough to read the result or headers, pico will hang.
httpc_connection_t settings;
ip_addr_t ip;
char apicall[1024];

int ConnectWifi(uint32_t country, const char *ssid, const char *pass, uint32_t auth)
{
	//if (cyw43_arch_init()) {
    if(cyw43_arch_init_with_country(country)){    
        //printf("WiFi init failed.\n");
        return 1;
    }

    cyw43_arch_enable_sta_mode();

    set_host_name("Pico_Monitor1");
    //printf("Wifi Connecting to: %s...\n",ssid);
    
    if (cyw43_arch_wifi_connect_timeout_ms(ssid, pass, auth, 10000)){        
        //printf("Wifi failed to connect to: %s\n",ssid);
        return 1;
    } 
    else{
        //printf("Wifi connected to: %s\n",ssid);
        LedOn();
        sleep_ms(500);
        LedOff();
        return 0;
    }
}

static void tcp_close_conn(struct tcp_pcb *pcb) {
    tcp_arg(pcb, NULL);
    tcp_sent(pcb, NULL);
    tcp_recv(pcb, NULL);
    tcp_close(pcb);
}

static err_t tcp_client_connected(void *arg, struct tcp_pcb *tpcb, err_t err) {
    if (err != ERR_OK) {
        //printf("Connection failed\n");
        tcp_close_conn(tpcb);
        return err;
    }

    //printf("Connected to server\n");

    sprintf(apicall, 
        "Monitor1,location=Greenhouse Temperature1=%3.3f\n"
        "Monitor1,location=Greenhouse Temperature2=%3.3f\n",
		Temperature1, Temperature2
        ); 

    //printf("strlen apicall: %i\n", strlen(apicall));
    //printf("apicall: %s|\n", apicall);

    char request[1024];
    snprintf(request, sizeof(request),
             "POST %s HTTP/1.1\r\n"
             "Host: %s\r\n"
             "Authorization: Bearer %s\r\n"
             "Content-Type: application/json\r\n"
             "Content-Length: %d\r\n\r\n"
             "%s",
             //POST_PATH, POST_URL, BEARER_TOKEN, strlen(post_data), post_data);
             POST_PATH, POST_URL, BEARER_TOKEN, strlen(apicall), apicall);

    err = tcp_write(tpcb, request, strlen(request), TCP_WRITE_FLAG_COPY);
    if (err != ERR_OK) {
        //printf("Failed to send data\n");
        tcp_close_conn(tpcb);
        return err;
    }
    else{
        //printf("Successfully sent data\n");
    }

    err = tcp_output(tpcb);
    if (err != ERR_OK) {
        //printf("Failed to flush data\n");
        tcp_close_conn(tpcb);
        return err;
    }

    return ERR_OK;
}

static err_t tcp_client_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *p, err_t err) {
    if (!p) {
        //printf("Connection closed by server\n");
        tcp_close_conn(tpcb);
        return ERR_OK;
    }

    //printf("Received data: %.*s\n", p->len, (char *)p->payload);
    pbuf_free(p);

    return ERR_OK;
}

void poll_sensors()
{
	Temperature1=10.5;
	Temperature2=21;
}

int main() {
    stdio_init_all();

    int status = ConnectWifi2(country, ssid, pass, auth);
    
    ip4addr_aton("192.168.1.1", &server_ip); // Use a function or DNS resolver to get the server IP
    //lwip_init();  //already called by cyw43_arch_init_with_country, calling it again will hang when calling sleep_ms()
    
    while(true) {        
        
        sleep_ms(100);
        
        
		//poll sensor data to get Temperature1 and Temperature2
		poll_sensors();

		//Post data

		pcb = tcp_new();
		if (!pcb) {
			printf("Error creating PCB\n");        
		}
		

		err_t err = tcp_connect(pcb, &server_ip, PORT, tcp_client_connected);
		if (err != ERR_OK) {
			printf("Error initiating connection\n");
			tcp_close_conn(pcb);
		}

		tcp_recv(pcb, tcp_client_recv);

        sleep_ms(30*1000);
        
#if PICO_CYW43_ARCH_POLL
        // if you are using pico_cyw43_arch_poll, then you must poll periodically from your
        // main loop (not from a timer) to check for WiFi driver or lwIP work that needs to be done.
        cyw43_arch_poll();
        sleep_ms(1);
#else
        // if you are not using pico_cyw43_arch_poll, then WiFI driver and lwIP work
        // is done via interrupt in the background. This sleep is just an example of some (blocking)
        // work you might be doing.
#endif        
    }

    cyw43_arch_deinit();
    return 0;
}
`

AdrianCX, 2025-08-15:
An example of a HTTPS POST to telegram: https://github.com/AdrianCX/pico_https_example/blob/main/https_request/main.cpp#L65

LWIP abstraction is here: https://github.com/AdrianCX/pico_https/blob/main/pico_tls/session.cpp

Can probably simplify if there's any interest. Right now using these for some home automations.

lavara123, 2025-08-16:
@AdrianCX 

Thank you very much for this library and examples.

With your code it does not connect to the normal home network wifi (router: keenetic). When I share wifi from my android phone it connects but the code below also connects to the normal home network. What do you think is the problem?

```c
bool connect_to_wifi_ssid() {
    if (ssid_pass) {
        ssid = cJSON_GetStringValue(cJSON_GetObjectItem(ssid_pass, "ssid"));
        pass = cJSON_GetStringValue(cJSON_GetObjectItem(ssid_pass, "pass"));
        
        if (cyw43_arch_init()) {
            printf("cyw43_arch init error\n");
            return false;
        }
        cyw43_arch_enable_sta_mode();
        printf("\n\n==========================\n"
            "Connecting to WiFi:%s\n"
                "==============================\n", ssid);
        if (cyw43_arch_wifi_connect_timeout_ms(ssid, pass, CYW43_AUTH_WPA2_AES_PSK, 10000)) { 
            printf("wifi sta connect error ssid:%s\n", ssid);
            cyw43_arch_deinit();
            return false;
        }
    } else {
        return false;
    }
    cJSON_Delete(ssid_pass);
    
    ip_addr_t addr = cyw43_state.netif->ip_addr;
    printf("connect successfully. get IP: %s\n", ipaddr_ntoa(&addr));
    return true;
}
```

I am working on the example in your code (https://github.com/AdrianCX/pico_https_example/tree/main/https_request)

I tried making the country my own country, but it still won't connect.
```c
cyw43_arch_init_with_country()
```

In your example, the unconnected part is only connected to the router in the home network
```c
if (cyw43_arch_init_with_country(CYW43_COUNTRY_NETHERLANDS)) {
        printf("failed to initialise\n");
        return 1;
    }    

    // Connect to wifi
    cyw43_arch_enable_sta_mode();
    cyw43_arch_disable_ap_mode();

    int err_code = cyw43_arch_wifi_connect_timeout_ms(WIFI_SSID, WIFI_PASSWORD, CYW43_AUTH_WPA2_AES_PSK, 30000);

    cyw43_wifi_pm(&cyw43_state, cyw43_pm_value(CYW43_NO_POWERSAVE_MODE, 2000, 1, 1, 1));

    if (err_code != 0)
    {
        for (int i=0;i<5;++i)
        {
            printf("[%d] failed to connect to wifi, ssid[%s], password[%s] err[%d], restarting in %d seconds\n", i, WIFI_SSID, WIFI_PASSWORD, err_code, 5-i);
            sleep_ms(1000);
        }

        #define AIRCR_Register (*((volatile uint32_t*)(PPB_BASE + 0x0ED0C)))
        AIRCR_Register = 0x5FA0004;
        return 1;
    }

```

PR #385 (2024-11-23): Add HTTP client examples
Examples to demonstrate HTTP, HTTPS and TLS validation Uses the lwip HTTP client

Fixes #318