raspberrypi/pico-examples #322: Pico nuke not working

YashasSingh, 2023-02-03:
When trying to use the pico nuke it seem to not restart and it rather thinks the file is needed. Check it out

https://user-images.githubusercontent.com/121969113/216627003-de95d07f-fe7e-4bbb-a857-399dc12576f8.mp4

peterharperuk, 2023-02-03:
What type and make of board do you have?

YashasSingh, 2023-02-03:
work now thank you very much

closed 2023-02-03 by peterharperuk

YashasSingh, 2023-02-03:
Never mind it does not work

aallan, 2023-02-03:
From your MP4 you've tried to drop the nuke.uf2 onto a Pico that is running CircuitPython, instead, unplug your Pico. Press and hold the BOOTSEL button, plug your Pico in while still holding the BOOTSEL button down. It will mount as a RPI-RP2 volume on your desktop, drag-and-drop your nuke.uf2 onto that volume, not your CircuitPython volume.