raspberrypi/pico-examples #323: Please make USB host more understandable

xployt404, 2023-02-03:
I've been trying to make my raspberry pi pico W read keyboard input from a keyboard that is attached to the micro USB port for a week now and I wasn't able to understand how to implement it into my code. Not only is the USB host stuff very badly documented, I also don't understand the examples nor how I can utilize the functions in my code. Please make the examples more understandable, document the functions better or publish an official tutorial.

peterharperuk, 2023-02-10:
Hi. You can read from the terminal with getchar_timeout_us. On the develop branch we have a new API to notify you when a character is available stdio_set_chars_available_callback.
Is that what you want or do you need to connect a USB keyboard directly?

xployt404, 2023-02-10:
> Hi. You can read from the terminal with getchar_timeout_us. On the develop branch we have a new API to notify you when a character is available stdio_set_chars_available_callback.
> Is that what you want or do you need to connect a USB keyboard directly?

Yes I want to plug the keyboard direcrly into the micro USB port

emillma, 2023-03-14:
> Hi. You can read from the terminal with getchar_timeout_us. On the develop branch we have a new API to notify you when a character is available stdio_set_chars_available_callback. Is that what you want or do you need to connect a USB keyboard directly?

Is there any documentation on how to use stdio_set_chars_available_callback? The feature looks nice, but I have not managed to get it working.

aallan, 2023-03-14:
> Is there any documentation on how to use stdio_set_chars_available_callback? The feature looks nice, but I have not managed to get it working.

See https://www.raspberrypi.com/documentation/pico-sdk/runtime.html#ga4b010bf88f65fc5d38a29d5538fa80e1

closed 2023-11-04 by xployt404