raspberrypi/pico-examples #328: Building Errors on master branch

dchithinh, 2023-02-11:
Hi everyone,

I just pulled the latest code from master branch. Tried build and there some errors in dma, i2c, etc. I fixed one by one to build what I want. I clone the repo long ago and build ok.
Here is one of the errors:
```[ 12%] Linking CXX executable sniff_crc.elf
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/sniff_crc.dir/sniff_crc.c.obj: in function `main':
sniff_crc.c:(.text.startup.main+0xce): undefined reference to `dma_sniffer_set_data_accumulator'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: sniff_crc.c:(.text.startup.main+0xd4): undefined reference to `dma_sniffer_set_output_reverse_enabled'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: sniff_crc.c:(.text.startup.main+0x118): undefined reference to `dma_sniffer_get_data_accumulator'
collect2: error: ld returned 1 exit status
gmake[2]: *** [dma/sniff_crc/CMakeFiles/sniff_crc.dir/build.make:823: dma/sniff_crc/sniff_crc.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:5528: dma/sniff_crc/CMakeFiles/sniff_crc.dir/all] Error 2
```
It seems the master branch was merged missing some functions.

lurch, 2023-02-11:
Those functions were recently [added](https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.0) to the master branch, so you probably just need to delete your `build` folder (and then create a new one) to get rid of your stale object-files?

dchithinh, 2023-02-11:
Yes, I did remove my build folder and try to re-configure build
```
cd pico-examples
sudo rm -r build
cmake -B build -S .
cmake --build build
```
Also, I tried to find dma_sniffer_set_data_accumulator() as the above error in the code but it is not there. The Readme.txt does not  give any library to be linked when build.

But the errors are still there.

Note: just report a errors, I am using that example at all.

Thank you.

dchithinh, 2023-02-11:
Oh, I take a look at your linked. It is my out of date SDK. 
Thanks for your help. 
Will close this thread.

closed 2023-02-11 by dchithinh

lurch, 2023-02-11:
> I just pulled the latest code from master branch.
...
But the errors are still there.

Ah, I didn't spot that this issue was in the `pico-examples` repo and not the `pico-sdk` repo, sorry for the confusion!