raspberrypi/pico-examples #339: clocked_input example clarification

hanetzer, 2023-03-05:
Hello. As part of a project I've been tinkering with, I've been playing with the idea of turning a raspi
pico into a LPC ([Low Pin Count](https://www.intel.com/content/dam/www/program/design/us/en/documents/low-pin-count-interface-specification.pdf)) sniffing device to synchronously sample the lpc bus signals, do some
light parsing and decoding with the data, and report that over usb or uart (were the usb interface fast
enough I'd just grab the raw data and process it in sigrok or similar but alas, 12mbps is far too slow
for the 33.3Mhz lpc clock).

Am I correct in reading the example that the pio samples in sync with the spi0 clock more or less
exactly?

lurch, 2023-03-05:
https://github.com/raspberrypi/pico-examples/blob/master/pio/clocked_input/clocked_input.pio says "The data is actually sampled one system clock cycle after the rising edge is observed, so a clock ratio of at least input_clk < clk_sys / 6 is recommended for good sampling alignment."

hanetzer, 2023-03-06:
ok, so if I'm understanding that right, ideally the system clock of the rpi pico should be greater than 6 times the target protocol clock? so for lpc, greater than 200mhz?

lurch, 2023-03-06:
Yeah, that's what it sounds like. Pinging @Wren6991 for clarification.

hanetzer, 2023-03-07:
and regarding the sm_config_set_in_shift bit, the threshold, is that per pin or overall?
I'm intending on reading 8 bits at a time (two are dummies but that's for alignment
purposes and overall logic easing).

I guess what I'm getting at is I just want to sample 8 (really 6) pins every observed
lpc clock cycle and be able to perform some simple logic on that byte on the c side
of things.

lurch, 2023-03-07:
> and regarding the sm_config_set_in_shift bit, the threshold, is that per pin or overall?

Looking at the API of that function https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#rpipf735077cf92a39c0da06 I'd assume it's per StateMachine rather than per-pin?