raspberrypi/pico-examples #341: Add an example for bidirectional (3-wire) SPI communication

mbartlett21, 2023-03-06:
For example, communicating with an SSD1677 eink display chip. ([Datasheet](https://www.waveshare.com/w/upload/2/2a/SSD1677_1.0.pdf))

Rather than having separate in/out data pins, they instead have a single SDA bidirectional pin, which depending on the command sent, is written to by either the display or the master. (SDI and SDO in the chart are actually a single pin)

[image]

peterharperuk, 2023-03-06:
In case it's useful, Pico W communicates with the wifi chip using SPI where MOSI and MISO are the same pin (also used as the interrupt line). It uses the pio to do this.

mbartlett21, 2023-03-06:
@peterharperuk 
So this would be the pertinent code?

https://github.com/raspberrypi/pico-sdk/blob/f396d05f8252d4670d4ea05c8b7ac938ef0cd381/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.pio#L20-L32

peterharperuk, 2023-03-06:
>any of the other PIO programs in that file used

No - they were used for bringup and are just kept for reference. Different devices might behave differently and the pio program might need tweaking. The code is in cyw43_spi_transfer and as written currently only supports a) write b) write then read, but that could be fixed I imagine.

matsobdev, 2024-07-17:
Hmmm, just crude idea. It might be possible with hardware SPI and tying Pico's MISO and MOSI with resistors (conecting Pico's MISO through a resistor to display data pin, and the same with Pico's MOSI).

Slion, 2024-10-06:
Looks like 3-wires half duplex SPI is very common for displays. Apparently those [Pimoroni Display Pack with ST7789](https://github.com/pimoroni/pimoroni-pico/issues/1004) are using it too. Having such an example would certainly benefit the community.

peardox, 2024-10-06:
The ST7789 uses the DC to signal Data/Command then and command issued determines the direction of the pin usually known as MOSI. Therefore when sending the command you'd have to set the direction of "MOSI" to the appropriate value in order to send/recieve data

lurch, 2024-10-07:
There's already [an example](https://github.com/raspberrypi/pico-examples/tree/develop/pio/st7789_lcd) for ST7789 :slightly_smiling_face:

peardox, 2024-10-07:
Yeah, trying to adapt it to do read ATM as read has useful features not exposed by FOSS libraries