raspberrypi/pico-examples #351: UDP Beacon does not work https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/udp_beacon/picow_udp_beacon.c

mytechnotalent, 2023-03-10:
https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/udp_beacon/picow_udp_beacon.c

The udp beacon does not function.  

I have, on the same network, a simple python UDP listener however nothing is being received.
```

import socket

UDP_PORT = 4444

# create a UDP socket
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

# bind the socket to a specific port
sock.bind(('0.0.0.0', UDP_PORT))

# continuously receive packets
while True:
    data, addr = sock.recvfrom(1024) # receive up to 1024 bytes
    print(f"Received packet from {addr}: {data.decode()}")
```
in the serial monitor I am sending packets such as
Sent packet 1
Sent packet 2
...
Sent packet 555

peterharperuk, 2023-03-10:
Hi. I might be wrong, but binding to '0.0.0.0' usually means you're binding to your IP address. The udp beacon example isn't sending to your address. By default its doing a broadcast. Try changing BEACON_TARGET in the code to your IP address and see if you start receiving something.

mytechnotalent, 2023-03-10:
@peterharperuk yes that worked thank you!

closed 2023-03-10 by mytechnotalent