raspberrypi/pico-examples #357: Skipping TinyUSB dual examples

approxdec, 2023-03-18:
g'day.
why - having done `git submodule update --init` for the sdk
do i get this? in particular the `TinyUSB available at` ... and 
but then `Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable`

this apprently makes \pico-examples\build\flash\program\flash_program.uf2
not work

regards
joe

I:\pico\pico-examples\build>cmake -G "NMake Makefiles" ..
Using PICO_SDK_PATH from environment ('I:\pico\pico-sdk')
PICO_SDK_PATH is I:/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from I:/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Python311/python.exe (found version "3.11.1") found components: Interpreter
TinyUSB available at I:/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at I:/pico/pico-sdk/lib/btstack
cyw43-driver available at I:/pico/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at I:/pico/pico-sdk/lib/lwip
mbedtls available at I:/pico/pico-sdk/lib/mbedtls
Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable
-- Configuring done
-- Generating done
-- Build files have been written to: I:/pico/pico-examples/build

kilograham, 2023-03-19:
> this apprently makes \pico-examples\build\flash\program\flash_program.uf2
not work

no reason why that should be. Pico-PIO-USB is a submodule of the TinyUSB submodule, so you have to initialize that separately (better than doing a full recursive init of TinyUSB which is huge)

cd into `lib/tinyusb` and do a `git submodule update --init hw/mcu/raspberry_pi/Pico-PIO-USB`

but then as I say this should have nothing to do with `flash_program`

kilograham, 2023-06-06:
I think this is solved

Sod-Almighty, 2023-06-14:
This is not solved. I got this same error today, and `git submodule update` didn't fix it :(

kilograham, 2023-06-14:
What compiler are u using; previously you had

```
The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
```

Right now, the Pico-PIO-USB code doesn't compile cleanly on GCC 11.3 or above. This will require a bump of both Pico-PIO-USB and TInyUSB to fix.

kilograham, 2023-06-14:
by solved, i meant the issue was known, but re-opening so we can track getting this fixed upstream (the GCC version issue)

Sod-Almighty, 2023-06-14:
```
# g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-alpine-linux-musl/12.2.1/lto-wrapper
Target: x86_64-alpine-linux-musl
Configured with: /home/buildozer/aports/main/gcc/src/gcc-12-20220924/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --build=x86_64-alpine-linux-musl --host=x86_64-alpine-linux-musl --target=x86_64-alpine-linux-musl --enable-checking=release --disable-fixed-point --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-werror --disable-symvers --enable-__cxa_atexit --enable-default-pie --enable-default-ssp --enable-languages=c,c++,d,objc,go,fortran,ada --disable-libssp --disable-libsanitizer --enable-shared --enable-threads --enable-tls --with-bugurl=https://gitlab.alpinelinux.org/alpine/aports/-/issues --with-system-zlib --with-linker-hash-style=gnu --with-pkgversion='Alpine 12.2.1_git20220924-r10'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.1 20220924 (Alpine 12.2.1_git20220924-r10)
```

I'm running on Alpine Linux, because I'm a masochist I guess. I had to install gcc and g++ explicitly.

kilograham, 2023-06-14:
yeah, i mean the workaround for now is to use GCC 11.2 or earlier

Sod-Almighty, 2023-06-14:
But GCC 12.2.1 _does_ compile the code. If you fix the bug in `src/pio_usb_device.c` that is.

kilograham, 2023-06-14:
yes, but that it is the project Pico-PIO-USB which is a dependency of TinyUSB, which is a dependency of the SDK... we can't just fix it; we have to wait for it to be fixed upstream.

Sod-Almighty, 2023-06-14:
Seems kinda an inefficient state of affairs. I wonder how anything gets done at all!

Personally, I'd clone the upstream repos, fix the problem, do a pull request – and if it wasn't actioned in a sensible amount of time, put the _working_ files into my project directly instead of having a dependancy.

Sod-Almighty, 2023-06-14:
At the very least, I'd put a note on the front page saying "edit this file, add this line and it'll work". You know, rather than the _current_ situation, where anyone running an operating system newer than a certain date is _completely screwed._

lurch, 2023-07-17:
> where can we track the issues upstream?

I'd assume you want to look at https://github.com/hathach/tinyusb/issues and https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues

kilograham, 2024-08-30:
fixed

closed 2024-08-30 by kilograham