raspberrypi/pico-examples #369: Can't compile tls_client example

kkimura1195, 2023-04-26:
On ubuntu 20.04, with a clean repo I cannot compile `picow_tls_client.c`. I have the error:

```
/pico-examples/pico_w/wifi/tls_client/picow_tls_client.c:61:93: error: 'TCP_WRITE_FLAG_COPY' undeclared (first use in this function)
   61 |     err = altcp_write(state->pcb, TLS_CLIENT_HTTP_REQUEST, strlen(TLS_CLIENT_HTTP_REQUEST), TCP_WRITE_FLAG_COPY);
      |                              
```

I have cloned this repository and I have launch compilation with `mkdir build && cd build && cmake -DPICO_BOARD=pico_w -DWIFI_SSID="network" -DWIFI_PASSWORD="passwd" .. && make`

peterharperuk, 2023-04-27:
It works for me. What branch are you using? Have you made sure both the pico-sdk and pico-examples are up to date?

lurch, 2023-05-04:
Have you done `git submodule update --init` inside your `pico-sdk` directory to make sure that the `lwip` submodule is up to date?

paulober, 2024-09-24:
@kkimura1195 Is this still an issue? Or has it been resolved?

peterharperuk, 2025-01-15:
No response. Closing.

closed 2025-01-15 by peterharperuk