raspberrypi/pico-examples #371: SPI: Only receiving first character?

fgolemo, 2023-04-30:
Heyhey folks,

I have 2 Picos, where one (master) has to send data to the other (slave) over SPI. The master is running Circuitpython and this is the code:

It just writes `(abcd)` every full second.

```python
import time
import board
import busio
from digitalio import DigitalInOut, Direction, Pull
from adafruit_bus_device.spi_device import SPIDevice

BAUD_RATE = 100_000
cs = DigitalInOut(board.GP17) 
cs.direction = Direction.OUTPUT

with busio.SPI(clock=board.GP18, MOSI=board.GP19) as spi_bus:
    spi_dev = SPIDevice(spi_bus, chip_select=cs, baudrate=BAUD_RATE)
    while True:
        with spi_dev as spi:
            spi.write(bytes("(abcd)", "ascii"))
        time.sleep(1)
``` 

And on the receiving end, I have to use C++ because SPI slave mode isn't implemented yet and my code looks like this:
It's supposed to read one character at a time from the MOSI line, so by repeated reading, it should get the whole message. 

```cpp
#include <stdio.h>
#include <string.h>
#include "pico/stdlib.h"
#include "pico/binary_info.h"
#include "hardware/spi.h"

#define PICO_DEFAULT_SPI_SCK_PIN 18 
#define PICO_DEFAULT_SPI_CSN_PIN 17
#define PICO_DEFAULT_SPI_RX_PIN 16

uint8_t received_buffer[1]; // receiving one character at a time

void setup() {
  Serial.begin(115200);
  
  spi_init(spi_default, 100 * 1000);
  spi_set_slave(spi_default, true);
  gpio_set_function(PICO_DEFAULT_SPI_RX_PIN, GPIO_FUNC_SPI);
  gpio_set_function(PICO_DEFAULT_SPI_SCK_PIN, GPIO_FUNC_SPI);
  // gpio_set_function(PICO_DEFAULT_SPI_TX_PIN, GPIO_FUNC_SPI); // pin not used, no MISO communication
  gpio_set_function(PICO_DEFAULT_SPI_CSN_PIN, GPIO_FUNC_SPI);
  bi_decl(bi_4pins_with_func(PICO_DEFAULT_SPI_RX_PIN, PICO_DEFAULT_SPI_TX_PIN, PICO_DEFAULT_SPI_SCK_PIN, PICO_DEFAULT_SPI_CSN_PIN, GPIO_FUNC_SPI));
}

void loop() {
  while(spi_is_readable(spi_default)) {
    spi_read_blocking(spi_default, 0, received_buffer, 1);
    Serial.print((char) received_buffer[0]);
    Serial.print(",");
  } 
}
```

...And I thought this is straight-forward enough, but when I run this, my logic analyzer shows me that the Master pi is sending correctly but the receiver pi only prints
`(,(,(,(,(,` and so on.

[image]

[image]

Does anybody have an idea why I'm only ever receiving the first character?
I've tried increasing the buffer to be size 6 but then I'll just get 6x `(` in my buffer.

peterharperuk, 2023-05-02:
What's value is returned by spi_read_blocking?

fgolemo, 2023-05-04:
@peterharperuk `spi_read_blocking` returns whatever I give it as the last parameter: 

- If I put in `1`, it reads one `(` and returns `1`
- If I put in `2`, it reads two `(` and returns `2`, etc.

(Also I updated my original question - I was originally using `spi_read_blocking`, not `spi_get_hw(spi_default)->dr`. I've only added that later after looking through the `pico-sdk/spi.c` code.