raspberrypi/pico-examples #383: cyw43 example access point with FreeRTOS

eamars, 2023-05-17:
I'm having trouble with merging below examples together: 
TCP server in poll/interrupt mode: https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/tcp_server/picow_tcp_server.c
IPerf with freertos: https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/freertos/iperf/picow_freertos_iperf.c

I'm looking to host a simple HTTP server with access point mode but running on RTOS. Is there any changes need to be made to port the TCP server to the FreeRTOS library? 

Kind Regards

peterharperuk, 2023-05-17:
I can't think what would need to change - probably a bug. Is there any reason why are you using the tcp_server example rather than the access point example?

eamars, 2023-05-17:
Hi @peterharperuk 
Thanks for your reply. I just updated the title and description for the issue. The cause was I changed the IP address for the host from 192.168.4.1 to 192.168.3.1. For some reason the DHCP server is still offering the 192.168.4.1 for the router address, according to the wireshark, therefore the host could not be resolved.

peterharperuk, 2023-05-17:
You changed the gateway address to 192.168.3.1 ad it still offers 192.168.4.1?

IP4_ADDR(ip_2_ip4(&state->gw), 192, 168, 4, 1);
IP4_ADDR(ip_2_ip4(&mask), 255, 255, 255, 0);

// Start the dhcp server
dhcp_server_t dhcp_server;
dhcp_server_init(&dhcp_server, &state->gw, &mask);

As far as I'm aware the dhcp server uses its own address and just changes the last number.

Do you have two netifs? If you have an AP and are also connected via STA mode - be aware of this bug https://github.com/raspberrypi/pico-examples/issues/382

eamars, 2023-05-17:
[image]

The new DHCP starts from packet 481. The 192.168.4.1 was offered, but the 192.168.3.16 was finally assigned to the client.

eamars, 2023-05-17:
@peterharperuk 
The only change I made was to change the subnet from 192.168.4.1 to 192.168.3.1. 

Only the AP mode is activated at the time of running the TCP server. Also please find my screenshot above regarding the DHCP packet.

peterharperuk, 2023-05-17:
If you changed the gateway address I don't see how it's possible for it to offer the wrong address.

eamars, 2023-05-18:
@peterharperuk  I finally trace back to the source of the 192.168.4.1
https://github.com/georgerobotics/cyw43-driver/blob/9bfca61173a94432839cd39210f1d1afdf602c42/src/cyw43_lwip.c#L182

For which when DHCP sends the offer, it uses 192.168.4.1 as the source. Unfortuantely due to the lack of knowledge I'm not sure why the system could not behave as normal when the IP address is changed to 192.168.3.1.

peterharperuk, 2023-05-18:
Ah right. Well done - I'd completely forgotten about that!
The latest driver (which should make it into the next release, has a macro CYW43_DEFAULT_IP_AP_GATEWAY to set this. I guess the dhcp server should use that.

peterharperuk, 2023-05-18:
> Is this coincidence for pico-example to also choose the 192.168.4.1 as the default subnet for example in the AP mode?

I think they have to match, right? The cyw43-driver comes originally from Micropython and I "borrowed" this DHCP server from Micropython, so I suspect that's why they match.

eamars, 2023-05-18:
Thank you very much!

kilograham, 2025-02-13:
merged into develop

PR #606 (2025-02-13): Stop hard coding IP gateway and mask
Fixes #383
commit b01d600f (2025-02-13): Stop hard coding IP gateway and mask (#606)

Fixes #383
closed 2025-02-13 by kilograham