raspberrypi/pico-examples #402: "hog_mouse_demo" can be connected to the computer but cannot control the movement of the mouse.

K1rin7, 2023-07-03:
I tried both "pico_w/bt/hid_mouse_demo" and "pico_w/bt/hog_mouse_demo", the former works well, but the latter does not.

### hid_mouse_demo:
I can use UART0 to control my mouse. (I enabled btstack debug)
[image]

### hog_mouse_demo:
- Can be detected and connected
- Cannot move the mouse

When I use UART0 to control the mouse, nothing happens (no mouse movement). 
[image]

I am using Clion and pico-sdk 1.5.1 to run the demo and I don't know how to solve this problem yet.

peterharperuk, 2023-07-04:
What system are you testing it with? I just tried it with a rpi4 running Linux and it works. There may be a problem with Windows.

K1rin7, 2023-07-04:
@peterharperuk I'm using windows 11, I just tested it with WSL(Ubuntu22.04), but the mouse didn't move either. I‘m now trying to test it with linux.

K1rin7, 2023-07-05:
I can connected to picow and moved the mouse using linux now.And it worked well with Ipad.

mringwal, 2023-07-06:
@K1rin7 does it now work with Windows 11 for you? Or only with Linux and iPad?

K1rin7, 2023-07-06:
@mringwal Only with Linux and iPad.

mringwal, 2023-07-29:
@K1rin7 please update BTstack to the latest version on the master branch or just copy the hog_mouse_demo.c from the master branch into the Pico SDK submodule into the examples folder and let me know if the mouse starts working on Windows for you, too.

K1rin7, 2023-07-29:
Hi, @mringwal 
Just now, I updated BTstack to the latest version (master and develop branch), but got the same result:

- picow doesn't work with Windows (W10 and W11)
- works well with Linux.

mringwal, 2023-07-29:
I forgot to push the change to master. Please try again and make sure that the "HID Mouse" was re-paired and/or try on a different Windows installation. I've tried earlier with Windows 10 in a VM and at least the buttons worked as expected (controlling mouse pointer is tricky as the mouse is controlled by the host environment)

K1rin7, 2023-07-29:
Hi, @mringwal
I tried "hog_mouse_demo" and "hog_keyboard_demo" on picow and now they both work well with Windows (both W10 and W11). 
Thanks!

mringwal, 2023-07-29:
Thanks for the confirmation.