raspberrypi/pico-examples #411: lwIP http request takes too long at altcp_connect() in TLS example

svetlollc, 2023-08-05:
## Problem

I successfully build and run a [TLS client Example][1] (`picow_tls_client_background` executable) from the GitHub, however a simple http GET request takes about 20 seconds. It does not actually depend on the server response: I tested server, that responds empty string at all, and server with about few kb output - no matter, it takes about 20 seconds to make a request.

As the result of debug, I understand, that 99% of the request time takes `altcp_connect`- before that, it takes several miliseconds, and after that, starting from the callback, i.e. `tls_client_connected`, it takes about millisecond until the request is made:

    Pico W turned on -- [few ms]  -- altcp_connect() -- [19 sec] -- tls_client_connected() -- [1 ms] -- I see comleted incoming request on the server

## Tried to solve

I made a reserch, I [found][2] that such behaviour, can be caused by lwIP configuration in `lwipopts.h`. It already sounds weird, because it probably should been tested by Raspberry Pi developers, who posted the code on GitHub.

So I set `#define MEMP_SANITY_CHECK 0`, set 

    #define LWIP_DEBUG 0 
    #define ALTCP_MBEDTLS_DEBUG  LWIP_DBG_OFF

I also tried to reduce `TCP_WND` or\and `TCP_MSS`, but any signficant changes produced cmake errors, that something overflows something. However, I reduced a bit that values, without errors at building.

However, all of that actions did not change request time - it still the same.

  [1]: https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/tls_client
  [2]: https://stackoverflow.com/a/15554408/19681078

svetlollc, 2023-08-05:
I also tried (although, as I wroted, this step is passed fast) to connect to ip directly, instead of DNS resolving by hostname, and it still takes same long.

svetlollc, 2023-08-05:
Also, in the code, I tried to set big (45) as well as small (1-5) values of `TLS_CLIENT_TIMEOUT_SECS` - it does not affect anything

svetlollc, 2023-08-06:
`picow_tls_client_poll` executable makes ~20 sec HTTP request as well.
Changing delay value in `cyw43_arch_wait_for_work_until(make_timeout_time_ms(1000));` does not solve the problem

svetlollc, 2023-08-21:
I also tried to call `altcp_nagle_disable`, when setup callbacks, tried to define `TPC_NODELAY` — no changes at all.

Can someone at least tell do they have or not same experience? Maybe it just how things works?

svetlollc, 2023-08-21:
So, in case of other people have at least few seconds long requests, after weeks of searching, I have only few possible issues:

1) In some examples, I saw that people somewhere defines country. In the Pico W examples, country doesn’t specify, maybe it is the reason.
2) Some examples call `ntp_time_init()` and `get_ntp_time()` — I do not know what it is, but maybe it relates to the problem, when you try to open some site on pc with wrong date, and browser doesn’t open it with error, that the clock is late.
3) Some examples call functions that setup mask, ip-adress, and similar stuff, as understand Pico W itself.
4) Low power supply, although I tried to connect Pico W to different sources: laptop usb, just a phone usb charger, other.

svetlollc, 2023-08-22:
Today I finally managed to make debug work, here is what Pico W outputs (nothing special):

    resolving <hostIspecified>
    DNS resolving complete
    connecting to server IP <123.456.7.8(ip that host pointing to)> port 443
    connected to server, sending request
    ***
    new data received from server:
    ***

    HTTP/1.1 200 OK
    Content-Length: 0
    Content-Type: text/html; charset=UTF-8
    Date: Tue, 22 Aug 2023 16:11:54 GMT
    Server: Apache
    X-Powered-By: PHP/7.0.32
    Connection: close

    connection was closed gracefully
    connection closed
    Test passed
    All done

* stuff with brackets <> is edited.

So, ofcouse 99% of time (about 18 seconds) takes `connecting to server IP <123.456.7.8(ip that host pointing to)> port 443` line

svetlollc, 2023-08-24:
Today I decided to run python, instead of C, to test http request on the Pico W, and it works fine (takes few seconds).

So, it is definetely me doing somehing wrong.

Can someone please at least point the direction, where should look to?

GoldeggCoder, 2024-11-18:
Anything new? I am facing the same issue with the tls example. I takes about 20 seconds per request.

Potzi123, 2024-11-22:
I stumbled upon the same problem as GoldeggCoder.

printf-hoven, 2025-01-08:
Same problem here too

peterharperuk, 2025-01-08:
There's unlikely to be a magic way to make https quick enough on pico. You need to keep the connection alive so you don't incur the setup cost every time you want to make a HTTP request.

peterharperuk, 2025-01-08:
Can you attach an example that does that?

GoldeggCoder, 2025-01-08:
Hmm...i used the basic example on the raspberry pi page with the urequests micropython package

GoldeggCoder, 2025-01-08:
`
import network
import requests

# Define ssid and password for connection
ssid = 'REPLACE_WITH_YOUR_SSID'
password = 'REPLACE_WITH_YOUR_PASSWORD'

# Setup WLAN and connect to WiFi
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(ssid, password)

# Make the HTTP GET Request
response = requests.get("https://www.google.com")

# Print response code and content
print('Response code: ', response.status_code)
print('Response content: ', response.content)
`

peterharperuk, 2025-01-08:
Thanks. I guess I should have known that! I'll have to figure out what we're doing wrong.

GoldeggCoder, 2025-01-08:
I appreciate your help👍

printf-hoven, 2025-01-09:
C++ code runs equally fast with "https://www.google.com", and more, such as "https://sound-au.com/". The python code, in any case, is calling into the underlying C functions only.

The issue of slow response occurs with some servers.

HOST = api.open-meteo.com. It takes 30 seconds to respond.

The same problem with HOST = katex.org

The problem seems NOT to be of the code, but of a slower TLS handshake communication between the Pico and [the "problematic"] remote HOSTs.  This problem doesn't occur on a PC browser because of higher CPU power, multi-cores and the richer set of ciphersuites available. Thanks peterharperuk for confirming my observations and giving sufficient inputs on this issue.

A randomly picked HOST shows this problem, which is a cause of concern.

GoldeggCoder, 2025-01-09:
I could differ from 3 seconds to 3 miniutes depending on how often the user wants to make a new measurement. If the module is shutdown it will send no requests.

peterharperuk, 2025-01-09:
If you're making a request every 3s, setting up a TLS connection for each request would seem like a bad idea.

GoldeggCoder, 2025-01-09:
Makes sense, I will try to setup the tls connection once and see if it is any faster.

svetlollc, 2025-01-09:
Hello, since people started pay some attention to the problem, in particularly, @printf-hoven provided some suggestions, today I have compiled again, test builds and reply.

1. http_client C example still takes very long.
2. Request to google.com via http_client C example indeed faster than some other - about 8 seconds vs 40.
3. Arguments about CPU power, however, wrong, since, after C program, I have ran a python script on the same Pico W board, and it takes 1 second to make a request to the same URLs (google.com, by the way some times does not work at all, for some reason).

```
import network
import requests

ssid = 'wifi name'
password = 'pass'

wlan = network.WLAN(network.STA_IF)
wlan.active(True)

wlan.connect(ssid, password)

requests.get("http://www.google.com") // takes about a second
```

svetlollc, 2025-01-10:
@peterharperuk 

Commenting `#define MBEDTLS_SHA256_SMALLER` does not change the behaviour. Also, again some hosts are doing faster: google.com, as mentioned earlier, and Your host from example  - only about 10 seconds from the C.

Python does fw-download-aliasl.raspberrypi.com for a second, but google.com can't do at all - after about 10 seconds it throws `OSError: [Errno 12] ENOMEM`.

peterharperuk, 2025-01-10:
Yes, the example is wasteful with memory.
I thought I'd check why www.google.com is 3 times quicker than raspberrypi.com. Google uses cipher suite TLS-ECDHE-ECDSA-WITH-AES-128-GC6 whereas raspberrypi.com is returning TLS-ECDHE-ECDSA-WITH-AES-256-GC4. Maybe that's why it's faster.

peterharperuk, 2025-01-10:
Not when I tested it. I'm measuring the time from the start of the request to the first header response.

svetlollc, 2025-01-10:
@peterharperuk I, of course also start measuring after wifi connection and other routine

svetlollc, 2025-01-10:
@peterharperuk I am not pretty familiar with networking, but I am near the pc currently, so can do some manipulations to the code, if You wish

printf-hoven, 2025-01-10:
If the following helps: 

microsoft.com, github.com are too slow.

atlas.microsoft.com never returns

peterharperuk, 2025-01-10:
No it doesn't

printf-hoven, 2025-01-10:
alttcp_connect doesn't return on requests to these hosts

microsoft.com, github.com, atlas.microsoft.com

the common thing to all of these is that they are hosted on a azure cloud.

This information can be used by the team that maintains pico-examples.

peterharperuk, 2025-01-10:
> alttcp_connect doesn't return on requests to these hosts

Like google, they use chunked encoding which won't work - the Lwip http client doesn't support it. There is also a memory leak in the code. It looks like internal_recv_fn should call the following. But I don't think that's causing the slow down.

altcp_recved(conn, p->tot_len);
pbuf_free(p);

printf-hoven, 2025-01-11:
@peterharperuk 

I can confirm that chunked encoding is NOT the issue. https://google.com is lightening fast on raspberry pico.

POINT 1:
=====

Following is the tls_client_recv callback for my C++ class. This code takes care of the memory release. It works. The code is an extract, but it should give the whole idea.

err_t CTCPClientSSL::tls_client_recv(void* arg, struct altcp_pcb* pcb, struct pbuf* p, err_t err)
{
  CTCPClientSSL* THIS = reinterpret_cast<CTCPClientSSL*>(arg);

  if (!p || (ERR_OK != err)) {

    if (ERR_OK == err)
    {
      THIS->_post_message(TCP_CLIENT_CONNECTION_CLOSED_BYREMOTE_HOST, THIS->_client_unique_id);
    }

    THIS->close_conn();

    return err;
  }

  if (p->tot_len > 0)
  {
    THIS->_tcp_handler(p);
  }

  // inform receipt
  altcp_recved(pcb, p->tot_len);

  pbuf_free(p);

  return err;
}

POINT 2
=====

The PROBLEM IS HERE: 

void CTCPClientSSL::connectAsync()
{
  cyw43_arch_lwip_begin();

---> THIS DOESNOT RETURN for azure cloud hosted API/sites, for some sites it takes TOO LONG, but for some it is lightening fast
  err_t err = altcp_connect(_client_pcb, &_remote_address, CTCPClientSSL::SSL_PORT, tls_client_connected);

  cyw43_arch_lwip_end();

  if (err != ERR_OK)
  {
#ifdef HDEBUG

    printf("CTCPClientSSL::connectAsync failed.ERRCODE = %d\n", err);

#endif

    _post_message(TCP_CLIENT_REQUEST_FAILED, _client_unique_id);

    close_conn();
  }
}

POINT 3
=====

My feeling is that some tweak is needed in the "mbedtls_config.h" file. I am using exactly the same lines as given in the pico-examples: https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/mbedtls_config_examples_common.h

peterharperuk, 2025-01-11:
I'm not sure it's anything to do with the mbedtls config. I suspect it's either the old version of mbedtls or a problem with the altcp code. I will write an example that does a TLS handshake directly with the mbedtls API to test things out.

printf-hoven, 2025-01-11:
Great! Possibly, the handshake is aborted by the remote server.

A err-rata correction to my earlier post.

Instead of 
======

---> THIS DOESNOT RETURN for azure cloud hosted API/sites, for some sites it takes TOO LONG, but for some it is lightening fast
err_t err = altcp_connect(_client_pcb, &_remote_address, CTCPClientSSL::SSL_PORT, tls_client_connected);

read this
=====

---> PicoW panics [some assertion failure?] here for azure cloud hosted API/sites, for some sites it takes TOO LONG, but for some it is lightening fast
err_t err = altcp_connect(_client_pcb, &_remote_address, CTCPClientSSL::SSL_PORT, tls_client_connected);

peterharperuk, 2025-01-15:
I think I might close this as it seems mbedtls is just a lot slower when build for debug. A more recent version of mbedtls is about a second faster so I might do something about that.

printf-hoven, 2025-01-16:
Yes! the issue can be closed now. It is settled.

Following remarks may be added.

POINT 1
=====
It helps to add these two lines to mbedtls_config 

// this does the trick mostly
// https://forums.raspberrypi.com/viewtopic.php?t=367740
#define MBEDTLS_KEY_EXCHANGE_ECDHE_RSA_ENABLED

// ?this one also?
#define MBEDTLS_SSL_TLS1_3_COMPATIBILITY_MODE 1

POINT 2
=====

There are no problems if a server responds with chunked response < 16kb each chunk.

However some sites like https://feeds.bbci.co.uk/news/world/rss.xml serve a blob of xml with Content-Length that exceeds 16kb, and leads to memory based connection closure, or error -13. For such cases the following can be tried. It worked perfectly for my use case.

// tweaking for particular use cases should be done and tested for side effects
#undef TCP_WND
#define TCP_WND  (16384 * 2)

The pico-examples can mention this fact for new users.

POINT 3
=====

Speed of response will be slower on Raspberry Pico due to reasons beyond our control. One reason could be the response delays specific to the remote server [in certificate checks, handshakes]. Another one has already been mentioned by @peterharperuk 

 > There's unlikely to be a magic way to make https quick enough on pico. 

The issue is now a candidate for closure. Great thanks to @svetlollc who flagged the issue, and ultimately led me to the solution.

eldiGH, 2025-01-17:
# MicroPython vs C TLS Benchmark Results

## TL;DR
- MicroPython is significantly faster than C for successful tls http requests. I'm unsure it's becouse of implementation or some mbedtls/lwip config options

I benchmarked MicroPython's `urllib.urequests` against a C implementation for TLS requests. Interestingly, the MicroPython version performs almost twice as fast in some cases. Below is a summary of the results and the link to the prepared repository for reference.

## Repository
The benchmark code is a slightly modified version of the TLS example. You can find it here: [TLS Benchmark Repo](https://github.com/eldiGH/pico-tls-benchmark).

## Remarks
- The inability to process all requests does not significantly impact the performance comparison, as the primary goal was to measure speed rather than handle all edge cases. I couldn't address these issues due to a lack of experience with MicroPython, but they could probably be resolved with more development effort.

---

## Benchmark Results

### C Implementation (Without `#define MBEDTLS_KEY_EXCHANGE_ECDHE_RSA_ENABLED` and `#define MBEDTLS_SSL_TLS1_3_COMPATIBILITY_MODE 1`)
```
Connected in 5551.58 ms 
Request to https://www.google.com/ SUCCEEDED in 3286.66 ms 
Request to https://www.youtube.com/ SUCCEEDED in 5071.41 ms 
Request to https://www.facebook.com/ SUCCEEDED in 5038.93 ms 
Request to https://www.wikipedia.org/ SUCCEEDED in 10238.48 ms 
Request to https://catfact.ninja/fact SUCCEEDED in 10474.11 ms
```

### C Implementation (With `#define`s set)
```
Connected in 5552.12 ms 
Request to https://www.google.com/ SUCCEEDED in 3267.37 ms 
Request to https://www.youtube.com/ SUCCEEDED in 5094.45 ms 
Request to https://www.facebook.com/ SUCCEEDED in 4893.79 ms 
Request to https://www.wikipedia.org/ SUCCEEDED in 10210.86 ms 
Request to https://catfact.ninja/fact SUCCEEDED in 10528.38 ms
```

### MicroPython Example
```
Connected in 5385 ms 
Request to https://www.google.com/ SUCCEEDED in 1755 ms 
Request to https://www.youtube.com/ FAILED in 2753 ms; exception: memory allocation failed, allocating 155136 bytes 
Request to https://www.facebook.com/ FAILED in 1449 ms; exception: Redirects not yet supported 
Request to https://www.wikipedia.org/ SUCCEEDED in 2919 ms 
Request to https://catfact.ninja/fact SUCCEEDED in 1358 ms
```

peterharperuk, 2025-01-17:
Yes. The more up to date version of mbedtls in micropython is quicker.

eldiGH, 2025-01-17:
Thank you so much for those branches, I was struggling a bit with the update. I hope they’ll be merged into pico-sdk and lwip someday. :D

eldiGH, 2025-01-18:
I conducted some additional tests. Updating mbedtls definitely improved the performance; however, it is still far from the response times achieved in MicroPython. I have no idea why TLS performs so much faster in MicroPython. Optimizing TLS performance in C is crucial for me, as it significantly impacts battery energy savings.

I added several URLs for testing and ensured they include different types of certificates. The results suggest that ECDSA certificates are significantly slower than RSA certificates, which is not surprising. However, in MicroPython, there seems to be no noticeable difference in performance between the two.

Here are the test results:

### C (mbedtls 3.5.1):
```
Request to https://www.google.com/ SUCCEEDED in 2618.14ms | altcp_connect time: 2385.84ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 4918.21ms | altcp_connect time: 2390.95ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 3739.06ms | altcp_connect time: 3088.90ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 4704.15ms | altcp_connect time: 3089.83ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 1632.81ms | altcp_connect time: 678.55ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 9380.63ms | altcp_connect time: 7233.62ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 6844.23ms | altcp_connect time: 5839.89ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 6089.02ms | altcp_connect time: 5744.63ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 6455.97ms | altcp_connect time: 5757.17ms | cert: ECDSA SHA-384
```

### MicroPython:
```
Request to https://www.google.com/ SUCCEEDED in 1301 ms | cert: RSA SHA-256
Request to https://www.youtube.com/ FAILED in 2593 ms; exception: memory allocation failed, allocating 146432 bytes | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 2522 ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ FAILED in 3312 ms; exception: memory allocation failed, allocating 146432 bytes | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 2525 ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 1279 ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 2435 ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1744 ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 2420 ms | cert: ECDSA SHA-384
```

### MicroPython (without reading response, so this should probably by only tls handshake time):
```
Request to https://www.google.com/ SUCCEEDED in 1205 ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 1244 ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 1288 ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 1419 ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 1182 ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 1332 ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 1470 ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1373 ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 1257 ms | cert: ECDSA SHA-384
```

svetlollc, 2025-01-18:
@eldiGH , thank You for the testing. In my case google.com also fails, that could be due to the difference of the response length because of the region (localisation) difference.

You also correctly mentioned, that even with newer mbedtls the C version far away from Python, while C should at least as fast, and actually even faster than script-language Python.

Since updating mbedtls solves the issue only a bit, I do not see reasons to close the issue, as @printf-hoven proposed.

In the last comments @printf-hoven and @peterharperuk seems to state, that https just could not be so fast in Pico W, due to its performance, while, as I, and @eldiGH state — it does not explain how it however could be fast with Python.

svetlollc, 2025-01-18:
A bit off-topic, but it is weird, that since Pico W release date in 2022 we seem to be the first, who discuss the issue with C implementation. 

Do everyone (including Raspberry Pi Pico W developers) just use Python so much ?

eldiGH, 2025-01-18:
> A bit off-topic, but it is weird, that since Pico W release date in 2022 we seem to be the first, who discuss the issue with C implementation.
> 
> Do everyone (including Raspberry Pi Pico W developers) just use Python so much ?

I had the same issues two years ago with SDK v1.5.0, but at that time, I didn’t have the time to finish the project. I switched to MicroPython and forgot about it. Now, seeing your issue here, I thought it would be helpful to point this out.

peterharperuk, 2025-01-18:
> Do everyone (including Raspberry Pi Pico W developers) just use Python so much ?

I've noticed that it's slow but never compared it to micropython. My bad. Upgrading mbedtls was not trivial as I've found out.

We need to use the same config as micropython and micropython doesn't use the alt TLS API, so there are more things to test

printf-hoven, 2025-01-19:
I have never done python/micropython - no idea - but I believe that it's functions should finally be a wrapper to some C library, like Java? 

If Micro-Python on Raspberry PicoW is not using TLS API, then which C library[?] could it be calling into? The source code needs to be seen.

peterharperuk, 2025-01-19:
It uses mbedtls directly. I'll post a link on Monday

eldiGH, 2025-01-21:
Although I lack experience with mbedTLS and lwIP, I wanted to investigate whether the performance issues are caused by the altcp API. To do this, I wrote some rather ugly code (apologies in advance!) that directly uses mbedTLS.

The results show a slight improvement in timing (except for LinkedIn), but overall, the performance is still far from what I achieved with MicroPython. Here are the measurements:

[Repository](https://github.com/eldiGH/pico-tls-benchmark/tree/altcp_replacement)

```
Request to http://www.google.com/ SUCCEEDED in 163.91ms | tcp_connect time: 16.75ms
Request to https://www.google.com/ SUCCEEDED in 2344.71ms (was 2618.14ms) | tcp_connect time: 16.02ms | tls_open_connection time: 2234.67ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 2391.72ms (was 4918.21ms) | tcp_connect time: 16.51ms | tls_open_connection time: 2231.50ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 4869.77ms (was 3739.06ms) | tcp_connect time: 19.31ms | tls_open_connection time: 3550.07ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 4064.81ms (was 4704.15ms) | tcp_connect time: 19.68ms | tls_open_connection time: 3544.80ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 9649.06ms (was 1632.81ms) | tcp_connect time: 19.15ms | tls_open_connection time: 9237.19ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 2402.35ms (was 9380.63ms) | tcp_connect time: 18.57ms | tls_open_connection time: 2220.42ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 2433.32ms (was 6844.23ms) | tcp_connect time: 38.55ms | tls_open_connection time: 2290.96ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 2439.95ms (was 6089.02ms) | tcp_connect time: 22.86ms | tls_open_connection time: 2222.02ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 2438.98ms (was 6455.97ms) | tcp_connect time: 17.28ms | tls_open_connection time: 2243.96ms | cert: ECDSA SHA-384

```

Perhaps there are some configuration options that could help speed things up further?

peterharperuk, 2025-01-21:
For what it's worth. The micropython mbedtls code is here https://github.com/micropython/micropython/blob/master/extmod/modtls_mbedtls.c#L523

The lwip config is here https://github.com/micropython/micropython/blob/master/ports/rp2/lwip_inc/lwipopts.h and the mbedtls config here https://github.com/micropython/micropython/blob/master/extmod/mbedtls/mbedtls_config_common.h

printf-hoven, 2025-01-21:
@eldiGH your code is same as shown in the pico-sdk examples? It calls into the same [problematic] function altcp_connect.

// https://github.com/eldiGH/pico-tls-benchmark/blob/altcp_replacement/c/tls.c

cyw43_arch_lwip_begin();
state->altcp_connect_start_time = time_us_64();
err = altcp_connect(state->pcb, &state->ip_addr, state->port, tls_client_connected);
cyw43_arch_lwip_end();

printf-hoven, 2025-01-21:
@peterharperuk The use of mbedtls sockets is what makes the difference. I wasn't able to use these sockets directly - tried the whole sunday, and gave up by today morning. 

Problem with the current lwip-mbedtls
=========================

The altcp layer runs into lock issues somewhere with respect to the wifi, which makes altcp_connect take longer, and the device hangs occasionally. [In my own case, I had to use watchdog timer, something that I personally would never like.]

The official pico lwip-ssl integration needs a serious examination by their team.

We cannot use alternatives like wolfssl till they find place in the official raspberry pico releases.

eldiGH, 2025-01-21:
> [@eldiGH](https://github.com/eldiGH) your code is same as shown in the pico-sdk examples? It calls into the same [problematic] function altcp_connect.
>
> // [...]

This code is from master branch. Branch that you've posted link to is not even containing altcp api include in the cmakelists

eldiGH, 2025-01-22:
I believe I've identified the cause of the performance issues. By comparing the mbedtls configuration in MicroPython with that in `pico_examples`, I found a single `#define` that significantly impacted performance: `MBEDTLS_ECP_NIST_OPTIM`. This optimization improves calculations involving prime numbers. I'll prepare a PR to `pico_examples` with this change. 

Additionally, updating the mbedtls version could further enhance performance, though the improvement is not as dramatic as with the NIST optimizations. Below are the benchmarks I prepared to illustrate the differences:

1. **mbedtls 3.5.1**  
   - **Direct mbedtls implementation**:  
```
Request to https://www.google.com/ SUCCEEDED in 992.15ms | tcp_connect time: 15.97ms | tls_open_connection time: 770.89ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 2805.15ms | tcp_connect time: 15.41ms | tls_open_connection time: 782.57ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 1232.77ms | tcp_connect time: 15.65ms | tls_open_connection time: 643.79ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 1870.86ms | tcp_connect time: 15.86ms | tls_open_connection time: 644.01ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 2482.85ms | tcp_connect time: 15.75ms | tls_open_connection time: 1539.83ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 914.65ms | tcp_connect time: 12.75ms | tls_open_connection time: 755.66ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 1225.90ms | tcp_connect time: 32.02ms | tls_open_connection time: 807.49ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1111.82ms | tcp_connect time: 35.18ms | tls_open_connection time: 795.49ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 1345.57ms | tcp_connect time: 14.20ms | tls_open_connection time: 752.01ms | cert: ECDSA SHA-384
```
   - **altcp API**:  
```
Request to https://www.google.com/ SUCCEEDED in 1529.10ms | altcp_connect time: 1301.41ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 3439.63ms | altcp_connect time: 1325.63ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 1712.38ms | altcp_connect time: 781.31ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 2641.68ms | altcp_connect time: 786.69ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 2558.97ms | altcp_connect time: 1720.11ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 2077.56ms | altcp_connect time: 1920.52ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 3899.78ms | altcp_connect time: 1673.26ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1954.08ms | altcp_connect time: 1668.77ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 2159.40ms | altcp_connect time: 1600.67ms | cert: ECDSA SHA-384
```

2. **mbedtls 2.28.8 (used in pico_sdk)**  
   - **Direct mbedtls implementation**:  
```
Request to https://www.google.com/ SUCCEEDED in 1144.05ms | tcp_connect time: 15.44ms | tls_open_connection time: 982.45ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 3018.50ms | tcp_connect time: 15.55ms | tls_open_connection time: 989.03ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 1604.67ms | tcp_connect time: 16.11ms | tls_open_connection time: 1007.83ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 2484.81ms | tcp_connect time: 15.91ms | tls_open_connection time: 1001.43ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 3310.16ms | tcp_connect time: 16.04ms | tls_open_connection time: 2263.56ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 1142.94ms | tcp_connect time: 13.11ms | tls_open_connection time: 967.36ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 1442.09ms | tcp_connect time: 33.07ms | tls_open_connection time: 1021.11ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1346.26ms | tcp_connect time: 14.37ms | tls_open_connection time: 959.37ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 1554.74ms | tcp_connect time: 14.53ms | tls_open_connection time: 960.47ms | cert: ECDSA SHA-384
```
   - **altcp API**:  
```
Request to https://www.google.com/ SUCCEEDED in 1577.83ms | altcp_connect time: 1374.81ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 3457.63ms | altcp_connect time: 1381.11ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 1712.29ms | altcp_connect time: 1071.59ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 2780.15ms | altcp_connect time: 1071.70ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 3237.63ms | altcp_connect time: 2359.84ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 2728.19ms | altcp_connect time: 2547.18ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 4302.00ms | altcp_connect time: 2148.68ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 2471.05ms | altcp_connect time: 2067.65ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 2666.43ms | altcp_connect time: 2081.96ms | cert: ECDSA SHA-384
```

[... cut ...]
Request to https://scotthelme.co.uk/ SUCCEEDED in 8348.97ms | altcp_connect time: 7980.87ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 8902.52ms | altcp_connect time: 8053.41ms | cert: ECDSA SHA-384
```

### Conclusions:
1. Adding NIST optimizations to mbedtls significantly improves performance.  
2. Updating the mbedtls version also enhances performance, though to a lesser extent.  
3. The altcp implementation in lwIP is suboptimal and has its own performance issues.

printf-hoven, 2025-01-22:
Great! @eldiGH ! Yes - it speeds up significantly. I tested it myself. 

I made a mistake in this post
==================

> [@eldiGH](https://github.com/eldiGH) your code is same as shown in the pico-sdk examples? It calls into the same [problematic] function altcp_connect.
>
> // [...]

eldiGH, 2025-01-22:
I've also conducted a benchmark for the Pico 2 W. The difference here is huge as well.

1. **altcp api mbedtls 2.28.2 (pico_examples case)**
```
Request to https://www.google.com/ SUCCEEDED in 1459.27ms | altcp_connect time: 1306.74ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 2321.17ms | altcp_connect time: 1310.00ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 2423.75ms | altcp_connect time: 2012.98ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 3110.54ms | altcp_connect time: 2009.97ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 6654.90ms | altcp_connect time: 5905.18ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 4921.75ms | altcp_connect time: 4766.72ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 4682.30ms | altcp_connect time: 3855.69ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 4161.62ms | altcp_connect time: 3897.14ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 4159.72ms | altcp_connect time: 3798.21ms | cert: ECDSA SHA-384
```

2. **altcp NIST mbedtls 2.28.8**
```
Request to https://www.google.com/ SUCCEEDED in 703.46ms | altcp_connect time: 563.27ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 1539.73ms | altcp_connect time: 570.12ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 1019.97ms | altcp_connect time: 536.25ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 1805.81ms | altcp_connect time: 542.88ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 1361.35ms | altcp_connect time: 724.42ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 1545.88ms | altcp_connect time: 1062.89ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 2730.91ms | altcp_connect time: 885.54ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1136.03ms | altcp_connect time: 893.86ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 1173.38ms | altcp_connect time: 826.92ms | cert: ECDSA SHA-384
```

3. **altcp NIST mbedtls 3.5.1**
```
Request to https://www.google.com/ SUCCEEDED in 662.66ms | altcp_connect time: 502.43ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 1471.97ms | altcp_connect time: 502.69ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 824.67ms | altcp_connect time: 395.35ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 1191.88ms | altcp_connect time: 396.87ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 1245.41ms | altcp_connect time: 536.72ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 1109.90ms | altcp_connect time: 900.27ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 2382.77ms | altcp_connect time: 803.88ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 1093.58ms | altcp_connect time: 741.32ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 1178.18ms | altcp_connect time: 736.35ms | cert: ECDSA SHA-384
```

4. **direct mbedtls 3.5.1 with NIST**
```
Request to https://www.google.com/ SUCCEEDED in 529.54ms | tcp_connect time: 15.37ms | tls_open_connection time: 408.24ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 1427.17ms | tcp_connect time: 15.29ms | tls_open_connection time: 417.19ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 871.29ms | tcp_connect time: 15.49ms | tls_open_connection time: 360.27ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 1403.52ms | tcp_connect time: 15.32ms | tls_open_connection time: 359.10ms | cert: RSA SHA-256
Request to https://www.linkedin.com/ SUCCEEDED in 1600.67ms | tcp_connect time: 15.78ms | tls_open_connection time: 606.37ms | cert: RSA SHA-256
Request to https://catfact.ninja/fact SUCCEEDED in 558.23ms | tcp_connect time: 12.43ms | tls_open_connection time: 394.44ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 739.02ms | tcp_connect time: 33.17ms | tls_open_connection time: 434.88ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 727.53ms | tcp_connect time: 13.57ms | tls_open_connection time: 397.02ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 788.35ms | tcp_connect time: 13.84ms | tls_open_connection time: 388.05ms | cert: ECDSA SHA-384
```

5. **direct mbedtls 3.5.1 with NIST RISC-V (potentially build with default compailer so potentially not optimized as mentioned [here](https://forums.raspberrypi.com/viewtopic.php?t=375268))**
```
Request to https://www.google.com/ SUCCEEDED in 603.44ms | tcp_connect time: 15.41ms | tls_open_connection time: 415.85ms | cert: RSA SHA-256
Request to https://www.youtube.com/ SUCCEEDED in 1575.53ms | tcp_connect time: 15.35ms | tls_open_connection time: 416.03ms | cert: RSA SHA-256
Request to https://www.facebook.com/ SUCCEEDED in 977.86ms | tcp_connect time: 15.57ms | tls_open_connection time: 492.32ms | cert: RSA SHA-256
Request to https://www.whatsapp.com/ SUCCEEDED in 1429.74ms | tcp_connect time: 15.74ms | tls_open_connection time: 364.70ms | cert: RSA SHA-256
[... cut ...]
Request to https://catfact.ninja/fact SUCCEEDED in 632.39ms | tcp_connect time: 12.63ms | tls_open_connection time: 388.14ms | cert: ECDSA SHA-256
Request to https://www.wikipedia.org/ SUCCEEDED in 748.35ms | tcp_connect time: 32.52ms | tls_open_connection time: 433.59ms | cert: ECDSA SHA-384
Request to https://scotthelme.co.uk/ SUCCEEDED in 822.08ms | tcp_connect time: 35.11ms | tls_open_connection time: 439.42ms | cert: ECDSA SHA-384
Request to https://stackoverflow.com/ SUCCEEDED in 807.25ms | tcp_connect time: 14.27ms | tls_open_connection time: 391.57ms | cert: ECDSA SHA-384
```

printf-hoven, 2025-01-25:
After spending some time, and tasting success, the following are my observations.

pico_lwip_mbedtls has problems - it is un-reliable both in speed and memory usage. Works only for responses that are below, say, 2kb. Chances of memory leakage are low, but problems surface when large memory allocations are required.

Adding NIST optimizations to mbedtls as suggested by @eldiGH increases the speed, but not reliability for use in a commercial application. [see #define MBEDTLS_ECP_NIST_OPTIM on https://siliconlabs.github.io/Gecko_SDK_Doc/mbedtls/html/group__sl__crypto__config.html#ga6c2a0ec24ee2e843114131f1087653cf]

The direct use of mbedtls as suggested by @eldiGH is a way forward. I have tested this code for a long duration, continuous run. Memory allocations are not required beyond a single span of 2048 bytes, so this approach works better. [See DECODED_DATA_BUF_SIZE in https://github.com/eldiGH/pico-tls-benchmark/blob/altcp_replacement/c/tls.c]

eldiGH, 2025-01-28:
I conducted additional tests regarding optimizations. Here are my findings:

## 1. NIST Curve Optimizations:
- Proven reliable in testing
- Only impact is increased binary size (465KB -> 469KB in my case)
- No reported issues found in documentation
- No problems encountered during testing

## 2. Hardware SHA256 acceleration on Pico 2 W:
I benchmarked hardware SHA256 acceleration and identified an issue: `mbedtls_tls_handshake` attempts to calculate 2 hashes simultaneously, which blocks the entire Pico with the current SDK implementation.

**Workaround attempted:**
Implemented caching of the first hash data and calculating it at a more appropriate moment

**Results:**
- Requests function normally
- Total request time increased by approximately 30-60ms on average

**Conclusion:** Hardware acceleration with the current implementation leads to performance degradation rather than improvement.

Hardware acceleration pico 2 branch: [link](https://github.com/eldiGH/pico-tls-benchmark/tree/pico-2-hardware-acceleration)

peterharperuk, 2025-01-28:
> mbedtls_tls_handshake attempts to calculate 2 hashes simultaneously

Is it mbedtls_entropy_func that does this or am I missing some other cases? Is that where you did the caching?

eldiGH, 2025-01-28:
I implemented caching for all data that gets passed to the mbedtls_sha256_update function. This means the current implementation in sha256_alt.c will only work for the first request made, since cache is not cleared. I'm sorry for that code, it is just PoC:

[Declaration](https://github.com/eldiGH/pico-tls-benchmark/blob/ff1daebca4d1c68d26a7a8c22ba690fc536b6e5e/c/sha256_alt.c#L26)
```c
uint8_t buf_0[8192] = {0};
size_t buf_0_len = 0;
```

[Update call](https://github.com/eldiGH/pico-tls-benchmark/blob/ff1daebca4d1c68d26a7a8c22ba690fc536b6e5e/c/sha256_alt.c#L53)
```c
    if (ilen == 0 || ctx->index == 0) {
        memcpy(buf_0 + buf_0_len, input, ilen);
        buf_0_len += ilen;
        return 0;
    }
```

[Finish call](https://github.com/eldiGH/pico-tls-benchmark/blob/ff1daebca4d1c68d26a7a8c22ba690fc536b6e5e/c/sha256_alt.c#L67)
```c
sha256_result_t result;

    if (ctx->index == 0) {
        pico_sha256_start_blocking(&ctx->pico_sha256_state, SHA256_BIG_ENDIAN,
                                   PICO_MBEDTLS_SHA256_ALT_USE_DMA);
        pico_sha256_update_blocking(&ctx->pico_sha256_state, buf_0, buf_0_len);
        pico_sha256_finish(&ctx->pico_sha256_state, &result);
        memcpy(output, result.bytes, 32);
        return 0;
    }
```

h1771t, 2025-04-19:
FYI.
My use case is to query google weather api (https://weather.googleapis.com/v1/forecast/days:lookup?key=)
I'm using adapted code from pico-examples http_client, api returns chunked json.
On first attempt, it would print part of the json response, then timeout and error.
After reading this thread and adding:

> #undef TCP_WND
> #define TCP_WND (16384 * 2)

It now is able to fetch the entire json response and return successfully.

printf-hoven, 2025-04-19:
The direct mbedtls code by @eldiGH = https://github.com/eldiGH/pico-tls-benchmark/tree/altcp_replacement is perfect on speed and reliability - I have been using [almost] this code for streaming audio over https, and also for newsfeeds from sites like bbc. The failure rate is zero or nearly zero.

It's better to bypass any by-passable wrappers, altcp in this case.

printf-hoven, 2025-04-19:
This is the C++ class [.h and .cpp file] that I wrote based on the above code. Making a repository is not possible for me because it is a part of a bigger shabbily written library highly specific to my own needs.

I am not sure if it will compile on VSCode setups, but it should give some idea - 

CTCPClientSSLMBEDDirect.h
// IDE Visual C++ used
==================

#pragma once
#include "lwip/tcp.h"
#include "lwip/pbuf.h"
#include <string>
#include <string_view>
#include "mbedtls/ssl.h"
#include "mbedtls/entropy.h"
#include "mbedtls/ctr_drbg.h"
#include "mbedtls/error.h"
#include "pico/util/queue.h"
#include "Base.h"

/// <summary>
/// the first arg is pbuf* for HTTP requests, and uint8_t* for HTTPS requests
/// the second is the length of the data
/// the third is the client unqiue id
/// fourth is true for HTTPS
/// fifth is void* custom data
/// return true to force close the connection
/// </summary>
typedef bool (*client_tcpssl_handler_blocking)(void* const, uint16_t, uint8_t, bool, void*);

/// <summary>
/// this class by-passes the altcp lwip-mbedtls library
/// it uses mbedtls directly, should be preferred
/// </summary>
class CTCPClientSSLMBEDDirect
{

  static constexpr uint16_t HTTPS_PORT = 443;

  static constexpr uint16_t HTTP_PORT = 80;

  static constexpr uint8_t POLL_TIME_S = 15;

  static constexpr uint16_t DECODED_DATA_BUFF_LEN = 2048;

public:

  // review _port == usages before adding new request types
  enum Connection_Type { HTTPS, HTTP };

public:

  CTCPClientSSLMBEDDirect(queue_t*, client_tcpssl_handler_blocking, void* = nullptr);

public:
  /// <summary>
    /// on failure, restart of device or re-creation of the object is required
    /// keep-alive = true returns ERR_OK on polling, otherwise connection is closed
    /// </summary>
  bool write_request_async(std::string_view host, std::string_view request, Connection_Type, uint8_t, uint16_t, int16_t max_response_length = -1, bool keep_alive = false);

  uint8_t get_client_id() { return _client_unique_id; }

  bool is_active() { return nullptr != _client_pcb; }

private:

  void* _custom_data;

  queue_t* _global_que;
[... 677 lines cut ...]
  else
  {

    err_t err = tcp_write(THIS->_client_pcb, THIS->_current_request.data(), THIS->_current_request.size(), TCP_WRITE_FLAG_COPY);

    if (err != ERR_OK) {

#ifdef HDEBUG

      printf("CTCPClientSSLMBEDDirect::write_request_async failed\n");

#endif

      THIS->_post_message(TCP_CLIENT_REQUEST_FAILED, THIS->_client_unique_id);

      return THIS->close_conn();

    }

    tcp_output(THIS->_client_pcb);

    return err;
  }

}

[2 more replies]