raspberrypi/pico-examples #413: flash_nuke for raspberry pi pico w

jorgechacblogspot, 2023-08-19:
In the following link:

https://github.com/raspberrypi/pico-examples/blob/master/flash/nuke/nuke.c

The necessary instructions to create the file <flash_nuke.uf2> to reset the raspberry pi pico board are described in C language, but in the pico w said led 25 lights up via wifi, how could this code be modified to work on a raspberry pi pico w?

lurch, 2023-08-20:
Have a look at https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/blink

mkerna, 2023-08-31:
Sorry, jorge, this is not a code issue.  The code correctly checks for a PICO_DEFAULT_LED_PIN.  The Pico W definition file correctly has no PICO_DEFAULT_LED_PIN.  So if you compile the code for the Pico W board it will work correctly.  If you don't specify a board during compile it will default to the standard Pico.  You may want to consult the documentation for how to compile for a specific board.  There also are forums at [https://forums.raspberrypi.com/viewforum.php?f=143](url) for any questions you may have.

Please close this issue.

mkerna, 2023-08-31:
Sorry, I think I misunderstood jorge's issue.  He would like the flash nuke firmware to flash the LED on the Pico W.  I still recommend that this is not an issue, as the flash nuke firmware would then require WiFi to flash the LED.  Flash nuke is meant to be quick and minimal.

jorgechacblogspot, 2023-08-31:
I solved the problem myself, it is a matter of defining PICO_DEFAULT_LED_PIN 25, which is disabled in the code and making it blink in pi pico w

#define PICO_DEFAULT_LED_PIN 25

Having done this and modified the code, the same pico sdk generates the .uf2 file that will be executed directly from the SRAM and the flash memory of the Raspberry Pi Pico W board will be erased, turning the Builtin LED on the board on and off 3 times to indicate that the program has finished successfully.

Apparently people do not understand that in the pico pi, the builtin led turns on easily by GP25, but in the pico w it turns on only by wifi in WL_GPIO0 through the CYW43439 module.

closed 2023-08-31 by jorgechacblogspot

junaidcodingmaster, 2025-03-27:
> Have a look at https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/blink

Ohhhh , OVER CLOCKING script.