raspberrypi/pico-examples #421: blink does not blink on a Pico W

repentsinner, 2023-09-17:
The Pico W 'hello world' (aka blink) out-of-the-box experience is terrible as of [sdk-1.5.1](https://github.com/raspberrypi/pico-examples/tree/sdk-1.5.1).

Please add at least an additional `#warning if using a pico w, blink has moved to pico-examples/pico_w/wifi/blink` in `pico-examples\blink\blink.c`. Preferred would be to update `blink.c` to include both the vanilla and W code paths. I suspect beginners to the Pico (W)(H) platform using the C SDK will be able to figure out what is going on, and it could be a good place to include some comments about other potential gotcha differences between the two boards.

While truly the LED is now connected to the wireless chip and therefore pedantically sample code for it should be in the `pico-examples/pico_w` directory, I highly doubt anyone looking for `blink` cares about that distinction when looking for a first thing to run to figure out if their board is working _at all_.

The official [overview documentation](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#raspberry-pi-pico-w-and-pico-wh) has a few W-specific notes; none of them describe the moved LED. It is only section 3.8 of [the Pico W datasheet](https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf) that gets around to rather dryly describing this move that then *might* point someone in the correct direction in the examples directory:

> There are three GPIO pins from the CYW43439 that are used for other board functions and can easily be accessed via the SDK:
> [...]
> WL_GPIO0
> OP connected to **user LED**

#350 has drifted off topic to somewhat address this, but it seems like a new issue is more appropriate.

lurch, 2023-09-17:
@aallan I've just had another look at Chapter 3 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and it has this boxout on page 9
[image]
but it's not very clear that the user actually needs to compile and build a **different** `blink` example program if they're using a Pico W board rather than a Pico board.

peterharperuk, 2023-09-18:
> Preferred would be to update blink.c to include both the vanilla and W code paths. 

Yes, I forget how we got into this situation. We should probably do something as it seems to be causing confusion.

XeYaRiQu, 2023-12-18:
@repentsinner Thank you for opening this issue as I am a beginner and have just experienced what you described. I followed with the "Getting Started with Raspberry Pi Pico" document and assumed that the Pico and Pico W were the same except for the addition of the WiFi/BT chip. Panicked for a moment thinking I had a defective Pico W.

aallan, 2023-12-18:
This will be fixed in the next documentation release, where the two blink examples are explicitly addressed in the _Getting Started_ book.

repentsinner, 2024-08-23:
Thanks so much @peterharperuk!

yoyojacky, 2024-09-04:
[image]
[image]
what should i do ? replace it with what kind of header file?

peterharperuk, 2024-09-04:
To include cyw43_arch.h you need to link to pico_cyw43_arch_none

lurch, 2024-09-04:
@yoyojacky `cyw43_arch.h` isn't in `pico-sdk/lib/cyw43-driver/src/` because it's actually in `pico-sdk/src/rp2_common/pico_cyw43_arch/include/pico/` :wink: 

If you're having problems compiling the `blink` example though, I suggest you create a new issue describing your problem in detail, rather than commenting on an already-closed issue.

PR #507 (2024-07-22): Make the blink example work on Pico W
It seems to confuse people that the blink example doesn't work on PicoW. To avoid further confusion, make it work, rather than build and silently do nothing.

Fixes #421