raspberrypi/pico-examples #423: I2C LIS3DH Sample Temperature

agatha2, 2023-09-30:
Concerning the [I2C LIS3DH sample](https://github.com/raspberrypi/pico-examples/tree/develop/i2c/lis3dh_i2c), the way the incidental temperature sensor is handled doesn't seem correct.

On [line 50](https://github.com/raspberrypi/pico-examples/blob/develop/i2c/lis3dh_i2c/lis3dh_i2c.c#L50), you're configuring with the query address `TEMP_CFG_REG`. However, it looks like that address ([`0xC0`](https://github.com/raspberrypi/pico-examples/blob/develop/i2c/lis3dh_i2c/lis3dh_i2c.c#L30)) was copy-pasted from the *value* you're giving it, also `0xC0`. According to [the datasheet (pg. 31)](https://www.digikey.com/htmldatasheets/production/734550/0/0/1/lis3dh-datasheet.html#pf1f), this is not a valid register<sup>[1]</sup>. That is quite bad, actually—the datasheet repeatedly warns against writing read-only or reserved registers, citing permanent hardware damage.

Assuming I'm right about all this (I don't yet have the hardware myself), the solution is simple: change the definition of `TEMP_CFG_REG` to `0x1f`, its actual address. <sup>[2]</sup>

---

<sup>[1]</sup> It may have alternately come from confusion with register `OUT_ADC3_L` at address *`0x0C`*, which is used (along with `0x0D` for >8-bit modes) for *reading out* the temperature (it's [used for that purpose correctly below](https://github.com/raspberrypi/pico-examples/blob/develop/i2c/lis3dh_i2c/lis3dh_i2c.c#L112)).

<sup>[2]</sup> Incidentally, while we're looking at the sample, I'd like some more documentation about it too. For example, I assume the scaling by `64` in `lis3dh_calc_value(⋯)` corrects for the chip's left-aligned data-output (16-bit -> 10-bit). The `sensitivity` must be found by realizing that the -2 to +2 default gee range is 4 gees over 10-bit (1024 code points), so 0.00390625 ≈ 0.004 gees/codepoint. These calculations are only valid for certain configurations, and they're wholly mysterious without diving into the datasheet.

lurch, 2023-10-01:
See #171 - this was an example written two years by an intern, and it's possible that the necessary hardware has been ~~lost~~ misplaced in the meantime. But if (when) you have working hardware and are able to verify the changes/improvements you're suggesting, a PR would be useful :+1:

peterharperuk, 2025-03-03:
I believe this is fixed

lurch, 2025-03-03:
Presumably fixed by #483