raspberrypi/pico-examples #428: Freertos wifi examples don't build with the latest version of FreeRTOS-Kernel

peterharperuk, 2023-10-04:
https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/ae3a498e435cecdb25b889f2740ea99027dd0cb1

kilograham, 2023-10-04:
OK, so this is a bit more involved; they have merged the `smp` branch into the `main` branch, so we will need to do some more investigation of if anything got broken (mixture of FreeRTOS cores, and SDK cores etc) for the next SDK release

raarts, 2024-01-04:
FreeRTOS 11.0 now defines FREE_RTOS_KERNEL_SMP instead of portSUPPORT_SMP.
Also, the examples don't run at all now, generating the following error:

```
assertion "get_core_num() == async_context_core_num(cyw43_async_context)" failed: file "/Users/raarts/pico/pico-sdk/src/rp2_common/pico_cyw43_driver/cyw43_driver.c", line 53, function: cyw43_irq_init
```

jdbruner, 2024-03-24:
It looks like FreeRTOS is looking for configNUMBER_OF_CORES but the SDK and example code is using configNUM_CORES. A quick workaround is to define them to be the same thing.

lurch, 2024-03-25:
> It looks like FreeRTOS is looking for `configNUMBER_OF_CORES` but the SDK and example code is using `configNUM_CORES`. A quick workaround is to define them to be the same thing.

See #429 :wink:

commit cac44988 (2024-05-02): Compilation issues with latest version of Freertos (#429)

Fixes #428
closed 2024-08-08 by kilograham