raspberrypi/pico-examples #430: Cannot Build Examples on Windows 10 WSL2 Ubuntu

RaphaelTetreault, 2023-10-11:
I have been trying to compile the Pico examples using Ubuntu in WSL2 on a Windows 10 machine. I can't seem to get it working despite starting from a clean distro image. I will outline the steps taken which I believe ought to work based on the resources I've read online.

```PowerShell
# We are in cmd.exe (Windows 10)
# Remove existing Ubuntu image
wsl --unregister ubuntu
# Install new Ubuntu image
wsl --install ubuntu
```
Image is installed, you create an account and password.
Now you are in the Ubuntu image.
```bash
# We are in Terminal (Ubuntu WSL)
# Prepare necessary components
sudo apt update
# This is from pico-sdk README
sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib

# You are in ~/
mkdir dev
cd dev
mkdir pico
cd pico
# You are in ~/dev/pico

# Clone Pico SDK and examples
git clone https://github.com/raspberrypi/pico-sdk --branch master --recursive
git clone https://github.com/raspberrypi/pico-examples --branch master

# Prepare variables for cmake/make
export PICO_SDK_PATH="~/dev/pico/pico-sdk"
export CXX="/usr/bin/gcc"

# Build
cd pico-examples
# You are in ~/dev/pico/pico-examples
mkdir build
cd build
# You are in ~/dev/pico/pico-examples/build
cmake ..
make
```
`make` will fail citing the compiler is broken. Installing `build-essential` will seemingly correct some aspect (perhaps something reliant on `g++`), but will then "fail spectacularly" by spewing a ton of warnings / errors to the terminal, all seemingly related to failed linking.

I've also tried tinkering with the specific compiler `CXX` points at, such as specifying `arm-none-eabi-gcc` specifically but errors are similar/persistent.

I've been at this for a few days now, and I have not been able to find answers.

lurch, 2023-10-12:
Have you tried doing this outside of WSL, following the Windows instructions in section 9.2 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?

RaphaelTetreault, 2023-10-12:
I did try the Windows native setup and it does work which is great. I realized after that I had piecemealed the instructions rather than using the provided document is a poor choice on my part.

So I decided to redo the steps in the document you provided in WSL and it did work. So it looks like some of my steps were causing the issue (probably setting CXX).

For others who might find themselves here, I recommend following the steps in [Getting Started with Raspberry Pi Pico (PDF)](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) Chapter 2.1 (the previously mentioned document by @lurch).

With the caveat that this might go outdated at any point, here are the instructions I followed which worked in Ubuntu in WSL.
```bash
# Navigate to where you wish to install the Pico SDK and Pico Examples
cd ~/
mkdir pico
cd pico

# Clone the Pico repositories
git clone https://github.com/raspberrypi/pico-sdk.git --branch master
cd pico-sdk
git submodule update --init
cd ..
git clone https://github.com/raspberrypi/pico-examples.git --branch master

# Update all relevant packages
sudo apt update
sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential libstdc++-arm-none-eabi-newlib

# Go to Pico Examples and prepare build
cd pico-examples
mkdir build
cd build

# Set environment variables (temporary) using relative path from ~/pico/pico-examples/build/
export PICO_SDK_PATH=../../pico-sdk

# Finally, build
cmake ..
make
```

RaphaelTetreault, 2023-10-12:
Thanks for the support, @lurch!

closed 2023-10-12 by RaphaelTetreault

aallan, 2023-10-12:
I guess folks are just too used to the official documentation being useless. 😫

RaphaelTetreault, 2023-10-12:
I think where I was lead astray was that I was expecting command-line instructions on setup in pico-examples the same as there was in `pico-sdk`. Looking back now, it is literally the first link in the examples README.

It might just be a user experience thing. I assumed consistency between READMEs for whatever reason. Perhaps a bit of indication as to where to look in the document would benefit?

I'm used to HTML documentation, either generated from source for API or written guides, so being dropped into a PDF and reading the content headings is not as searchable as web content.

lurch, 2023-10-13:
> I'm used to HTML documentation, either generated from source for API

We have API documentation (in HTML format) at https://www.raspberrypi.com/documentation/pico-sdk/