raspberrypi/pico-examples #437: examples/pico_w/wifi/tcp_client/CMakeLists.txt needs BOTH env variable and cmake define of TEST_TCP_SERVER_IP

dragonflight, 2023-11-24:
It may be just me, but it seems counter-intuitive and unlike other parameters to require both an environment variable and a cmake define to set the IP address

peterharperuk, 2023-11-24:
I don't think it does. I can't see it pulling TEST_TCP_SERVER_IP  from the environment. This needs to be set somewhere so the client knows which server to talk to.

dragonflight, 2023-11-24:
You are correct, I am new to cmake and misread the line
        TEST_TCP_SERVER_IP=\"${TEST_TCP_SERVER_IP}\"
to mean retrieving from the environment as opposed to putting quotes around the argument.

On the other hand as everything else ssid, password ... is passed through the environment, I did make the initial assumption (I would claim reasonably), that TEST_TCP_SERVER_IP was passed through the environment as well.

Probably a matter of taste.. 
Sorry, for creating a non-issue

closed 2023-11-24 by dragonflight