raspberrypi/pico-examples #439: [Question]: Pico gpio/dht_sensor Example not working

hasenradball, 2023-11-28:
Dear Pico Fellows,

the DHT Sensor example does not work.
I get always bad data.
It seems the code is written not robust enough, and also less comments.
Is it fine for you when I provide an improved code on basis on this:
[AM2302-sensor](https://github.com/hasenradball/AM2302-Sensor)

With this approach you will be also possible if the error is caused by a timeout or checksum_error.

lurch, 2023-11-29:
It's possible that https://github.com/raspberrypi/pico-examples/issues/423#issuecomment-1742244505 also applies here?

hasenradball, 2023-11-29:
@lurch thank for you comment.
I have additionally 2 questions (e.g.: for the dht example)

1) I generally see no usage of classes lets say, all code parts mainly placed in main.c files 
   For example providing an dht class via include, create an object and do it object orientated.
   Doe this not work with the pico c SDK?
   
2) I saw it mainly in the code of dht -> function `read_from_dht()`
  no line of comment in the whole file
  No clue whats the intention of this part if you look in lines 67 ... 71.
  Also no clue why the check if `humidity > 100 ` and `temperature >125` is there?
  If this occurs the data received is wrong.
  The sensor will never send such values.
  So if this happens it seems the data query procedure is wrong.

peterharperuk, 2023-11-29:
All the examples stick to C because that's just keeps things simple. Yes there seems to be a quite a few issues with this example. I doubt anyone has looked at it for some time. The points you raise are valid. Feel free to improve it and add some comments - or else I'll look at getting one of those sensors and doing it myself.

hasenradball, 2023-11-29:
Hi @peterharperuk,

thanks for your feedback I started to have a look at it.
I currently reworked a DHT Class ( see here: [AM2302-sensor](https://github.com/hasenradball/AM2302-Sensor) )

So I know what happens and what should happen.
I will check, but having a reviewer and discussion partner would be great!
:-)

hasenradball, 2023-11-29:
@lurch @peterharperuk 

I proposed a PR #440.
This code is adapted to the Pico c files and was mainly taken from here:
[AM2302-sensor](https://github.com/hasenradball/AM2302-Sensor)

peterharperuk, 2024-04-19:
I've used the lwip http client. There's an example here...

https://github.com/raspberrypi/pico-examples/pull/385

I think this depends on the following PR's so it might not compile on its own...
https://github.com/raspberrypi/pico-examples/pull/475
https://github.com/raspberrypi/pico-examples/pull/460

(I will get around to merging your PR's soon)

hasenradball, 2024-04-19:
Thank for fast reply I will have an look on it.

Regarding the PR's,here some nice feedback for the DHT sensor lib which was the basis for the Pico Example update:
Just for you information:
[https://github.com/hasenradball/AM2302-Sensor/issues/10](https://github.com/hasenradball/AM2302-Sensor/issues/10)

hasenradball, 2024-05-28:
@peterharperuk 
Hi Peter,

do you have some hint about my PRs, when to lokk at?
They are discussed a long time ago.

hasenradball, 2024-08-25:
@lurch @peterharperuk,

any interests to merge this?

lurch, 2024-08-26:
We've been a bit busy with [other things](https://www.raspberrypi.com/news/raspberry-pi-pico-2-our-new-5-microcontroller-board-on-sale-now/) recently :wink: