raspberrypi/pico-examples #446: No red LED lights up in apa102.c example program.

olibel270, 2023-12-09:
When running the apa102.c example program on a LED board, the "wave" pattern works flawlessly, but the color red never lights up.

Upon further investigation of the code, I believe I found the following typo in the main function:

```
      for (int i = 0; i < N_LEDS; ++i) {
            put_rgb888(pio, sm,
                       wave_table[(i + t) % TABLE_SIZE],
                       wave_table[(2 * i + 3 * 2) % TABLE_SIZE],
                       wave_table[(3 * i + 4 * t) % TABLE_SIZE]
            );
        }
```

Should rather be:

```
        for (int i = 0; i < N_LEDS; ++i) {
            put_rgb888(pio, sm,
                       wave_table[(i + t) % TABLE_SIZE],
                       wave_table[(2 * i + 3 * t) % TABLE_SIZE],
                       wave_table[(3 * i + 4 * t) % TABLE_SIZE]
            );
        }
```

For clarity, in the second wave_table[] access, the current code uses constant 2, rather than the likely intended variable t. 

Testing proves to fix the issue.

lurch, 2023-12-10:
> Testing proves to fix the issue.

Would you like to submit a PR with the fix? (against the `develop` branch)

olibel270, 2023-12-10:
For sure! I'll get on it.

lurch, 2025-02-14:
Fix merged into develop

closed 2025-02-14 by lurch