raspberrypi/pico-examples #454: [Question] Basic gatt ble server on Pico W

thewh1teagle, 2023-12-29:
I'm trying to find basic example of gatt ble server on Pico W with two way communication for single client.
can you point me to the right one?
Also, the pico-w/bt folder looks pretty empty. most of the folders have only CMakeLists.txt, why?

lurch, 2023-12-30:
> Also, the pico-w/bt folder looks pretty empty. most of the folders have only CMakeLists.txt, why?

As it says at https://github.com/raspberrypi/pico-examples/blob/master/README.md#pico-w-bluetooth
"They are examples from the Blue Kitchen Bluetooth stack, see [here](https://bluekitchen-gmbh.com/btstack/#examples/examples/index.html) for a full description."
And looking at https://github.com/raspberrypi/pico-examples/blob/master/pico_w/bt/CMakeLists.txt it seems like it pulls in the BT examples from `${PICO_SDK_PATH}/lib/btstack/example/`.

(I'm afraid I can't provide help myself with your other questions, as I know almost nothing about Bluetooth)

sonnny, 2023-12-30:
I have several ble examples (some might even have youtube demo) at below github link, source files are provided, there's even a bluetooth classic (not ble).

[https://github.com/sonnny](https://github.com/sonnny)

peterharperuk, 2024-01-22:
try pico-examples\pico_w\bt\standalone

closed 2024-01-22 by peterharperuk