raspberrypi/pico-examples #456: LCD1602_I2C EXample does not work

hasenradball, 2023-12-31:
Hello,

The example LCD1602_i2c does not work.
The code from the init function differs also from a working library.
So the init function from the working library is different.

lurch, 2024-01-02:
Looks like you need to make sure you change the `PICO_DEFAULT_I2C` define _before_ you include `i2c.h` ? (see https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_i2c/include/hardware/i2c.h#L76 )

Also, I believe if you `#define`d your new values before including any of the pico header-files, then you wouldn't need to `#undef` them.

hasenradball, 2024-01-02:
@lurch Hi,

thanks for the fast reply.
Ok this could be the point.
I will check this.

by the way a happy new year
😊

hasenradball, 2024-01-03:
@lurch 
I tried your proposal but does not help here.
It seems to be tricky to use the `i2c1`.
Is there an official example how to use `i2c1`?

lurch, 2024-01-03:
If you want to explicitly use `i2c1`, it might be easier to modify the code to explicitly refer to I2C1 and the new pins you're using, rather than trying to hackily override the "defaults" mechanisms?

lurch, 2024-01-03:
See https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#pinout-and-design-files and be careful that you don't mix up "pin numbers" and "GPIO numbers". Software will always use the GPIO numbers.

hasenradball, 2024-01-03:
@lurch 
I used always GPIO numbers.

hasenradball, 2024-01-03:
I just want to switch between these two ports:

[image]

lurch, 2024-01-04:
This is beginning to feel more like a support question, rather than a bug report. Support questions belong in https://forums.raspberrypi.com/

hasenradball, 2024-01-04:
@lurch I thik I had to double check this with a new Pico.

peterharperuk, 2024-01-16:
@hasenradball any chance you could tell me what components you're using to test this? Thanks.

hasenradball, 2024-01-16:
Hi Peter,

I had doupt that my pico works fine, I am waiting for my new layoutet pico dev board. then I will test again.
Then I will give some feedback.

Frank

hasenradball, 2024-01-16:
@peterharperuk 
Hi Peter what about the issue of the dht Sensor?
I was wonderin why you don‘ t merged my proposal into develop due to the fact it works, against the actual example don‘ t.

Frank

peterharperuk, 2024-01-16:
Yes. It'll get merged soon. It just needs me to test it again

hasenradball, 2024-01-17:
@peterharperuk 
are these boards not lovely?

[image]

peterharperuk, 2024-01-18:
I finally found a board to test this and it works fine on i2c1. I have pushed a fix that might make it a bit easier to change. The code is pretty bad - it would be nice to fix that sometime and get it to work with more modern devices.

P.S. The boards look very nice. What do they do?

closed 2024-01-18 by peterharperuk

lurch, 2024-01-18:
> The code is pretty bad

IIRC it was written by an intern, and everyone was so busy that no-one had the time to polish it.

hasenradball, 2024-01-18:
@peterharperuk i could not see any change in the examples.

peterharperuk, 2024-01-19:
I was referring to this commit https://github.com/peterharperuk/pico-examples/commit/3faa5f4a83de616c53d87767d41fff478ffaeeee

lurch, 2024-01-19:
> I was referring to this commit [peterharperuk@3faa5f4](https://github.com/peterharperuk/pico-examples/commit/3faa5f4a83de616c53d87767d41fff478ffaeeee)

That commit bypasses the "default pins" mechanism, which I guess means that it "breaks" that example for anybody using a board that redefines I2C onto a different set of pins? e.g. https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/sparkfun_thingplus.h#L46

peterharperuk, 2024-01-19:
It was just for testing so I could close this issue

hasenradball, 2024-02-01:
@peterharperuk @lurch 
Hi guys I could test now the LCD_1602 example with a new Pico.
I wrote a object oriented example to easiliy swith between i2c0 and i2c1.
It works now fine so the issue was located in damaged Pico.
Here connected to GPIO 2 and GPIO3

[image]

So issue could be closed.
Sorry for miss leading.

PS: why don't you use more object oriented examples, this would be much easier to adapt like in this to switch from i2c0 to i2c1.

Best regards
frank

hasenradball, 2024-02-01:
> I finally found a display to test this and it works fine on i2c1. I have pushed a fix that might make it a bit easier to change. The code is pretty bad - it would be nice to fix that sometime and get it to work with more modern devices.
> 
> P.S. The boards look very nice. What do they do?

Why don't you pushed the fix on the develop branch?
It seems to me it takes very long time if the examples (develop/master) branch are updated, isn't it?

Best regards

Frank

peterharperuk, 2024-02-01:
It's just not the top priority at the moment. We generally push as many changes as we can before making an sdk release. So it'll happen then or when it gets to the top of my todo list.