raspberrypi/pico-examples #457: PicoW Peripheral Secure Connection

mytechnotalent, 2024-01-01:
Can you please add a PicoW Peripheral Secure Connection example as the only ones available are plain text that can be seen by any sniffer.

lurch, 2024-01-01:
I'm not sure exactly what you're asking for, but are you looking for https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/tls_client ? (which was added in #305 )

peterharperuk, 2024-01-01:
I think he's referring to pairing in Bluetooth

mytechnotalent, 2024-01-01:
Yes @peterharperuk is correct I am referring to pairing in BLE I would love an example of such as the TLS example is great for WiFi but a BLE pairing example would be ideal and necessary for any real Bluetooth dev on the PicoW thank you in advance!

matsobdev, 2024-02-28:
There is [picow_bt_example_sm_pairing_peripheral](https://github.com/bluekitchen/btstack/tree/master/example/sm_pairing_peripheral.c) with corresponding `.gatt` file to encrypt characteristic data as well.

mytechnotalent, 2024-02-28:
Thank you @matsobdev I was hoping for something within the current RPI stack as this has licensing requirements.

lurch, 2024-02-28:
> this has licensing requirements.

See https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_btstack/LICENSE.RP