raspberrypi/pico-examples #462: Flashing pico fails

aredey, 2024-01-17:
I'm trying to flash a new pico with pico-hid.uf2 from a linux desktop, following the published procedure (pressing the BOOTSEL button while plugging in the micro USB power and then dropping (or cp-ing) the above firmware) and the process never completes, it starts the copy and then the pico "disk" disconnects without any error message. When trying to reconnect the uf2 file is not in the pico file system. Haven't found any solutions or how to troubleshoot the error  in the docs (here) or forums for a couple of days. Any pointers/solution would be appreciated. The desktop is a PopOS 22.04 machine but same issu trying to use Win10 for copying.

peterharperuk, 2024-01-17:
Have you tried other uf2 files?

lurch, 2024-01-17:
> When trying to reconnect the uf2 file is not in the pico file system.

That's to be expected. The filesystem that appears when you boot your Pico into BOOTSEL mode is only a "virtual" filesystem, and will _always_ appear "empty", even after copying a UF2 file to it. When the Pico disk disconnects, it means that it has finished loading the contents of your UF2 file into its internal flash memory, and it has rebooted to start running your application. 

See https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf for more detailed instructions.

aredey, 2024-01-17:
Thank you for the responses! I toyed with the idea of "nuking" the firmware but didn't actually try it so I haven't tried another uf2 file.

Thanks for the reference to the to the getting started doc, I'll dig in. I borrowed a Windows machine and when the pico "rebooted" it appeared to be emulating HUD functionality (like the system manager or whatnot showed the mouse/keyboard page - sorry, I'm not really a windows person) but Linux just disconnected the pico-drive silently, not sure how to figure out if the new firmware took. Will have a look at the Getting Started Guide, many thanks! (it would be good if this was mentioned or referred to in the [front page](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html). Many thanks Both, for the pointers!

closed 2024-01-17 by peterharperuk

lurch, 2024-01-17:
> Thank you for the responses! I toyed with the idea of "nuking" the firmware but didn't actually try it so I haven't tried another uf2 file.

There's no need to "nuke" your firmware before flashing a new UF2 - every time you copy a new UF2 onto a Pico in BOOTSEL mode it'll just overwrite whatever's in flash. The "nuke" UF2 file is just for ensuring that the entire flash memory is totally empty (should you need to do that for some reason).

> not sure how to figure out if the new firmware took.

You could try flashing the "blink" example to your Pico, which blinks the on-board LED on and off. Also, the flash memory on Pico is persistent across power-downs / unpluggings, so no need to write your firmware with Windows and _also_ with Linux :wink: 

> Will have a look at the Getting Started Guide, many thanks! (it would be good if this was mentioned or referred to in the [front page](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html).

It is https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#documentation