raspberrypi/pico-examples #474: stdio_usb input available callback can't call other stdio functions, and pico-examples iperf assumes it can.

arg08, 2024-02-24:
https://github.com/raspberrypi/pico-examples/blob/eca13acf57916a0bd5961028314006983894fc84/pico_w/wifi/iperf/picow_iperf.c#L50

For stdio_usb, the callback registered with  stdio_set_chars_available_callback() is called with the stdio_usb_mutex locked.  This means that such callbacks can't call any other stdio functions (they attempt to claim the mutex and fail).  In particular, this means you can't call getchar_timeout_ms() inside the callback.

This actually rather a dire limitation on the stdio callback mechanism, but that's how it is at the moment, and it means this piece of code in pico-examples doesn't work.

Fortunately, in this case the resolution is easy because it's already using async context: the stdio callback needs to do async_context_set_work_pending() unconditionally, and then key_pressed_worker_func() can read the key value and act on it accordingly.

peterharperuk, 2024-02-26:
Will be fixed by https://github.com/raspberrypi/pico-examples/issues/461

arg08, 2024-02-26:
True it's really a stdio_usb bug, but here it could trivially be fixed to make the example work with the existing stdio_usb.

I've commented over on https://github.com/raspberrypi/pico-examples/issues/1603 with a suggested fix for the underlying issue.

(but I've just noticed you've closed 1603 which was "there's a bug", but left open 461 which is "must provide an example for this feature" where the commentary says that actually it's a bug!).

peterharperuk, 2024-02-26:
I've reopened it

peterharperuk, 2024-08-19:
I believe this was fixed as described and the example was updated https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/iperf/picow_iperf.c#L40

closed 2024-08-19 by peterharperuk