raspberrypi/pico-examples #478: UART RX  Garbage data recived

engrzamanimran, 2024-03-02:
I am working on a recieving example but I recive the garbage data.
``#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/uart.h"
#include "hardware/irq.h"
#include "hardware/rtc.h"
#include "pico/util/datetime.h"

#define UART_ID uart0
#define BAUD_RATE 115200
#define UART_TX_PIN 0
#define UART_RX_PIN 1
#define DATA_BITS 8
#define STOP_BITS 1
#define PARITY    UART_PARITY_NONE
#define UART_BUFFER_SIZE 32

char buf[UART_BUFFER_SIZE];
uint32_t rx_buf_index = 0;

void on_uart_rx();

int main() {
    stdio_init_all();
    char datetime_buf[256];
    char *datetime_str = &datetime_buf[0];
    datetime_t t = {
        .year = 2024,
        .month = 03,
        .day = 01,
        .dotw = 5,
        .hour = 15,
        .min = 45,
        .sec = 00
    };
    rtc_init();
    rtc_set_datetime(&t);
    sleep_us(64);
    const uint LED_PIN = 25;
    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);
    gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
    gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);
    uart_init(UART_ID, 2400);
    int __unused actual = uart_set_baudrate(UART_ID, BAUD_RATE);
    uart_set_hw_flow(UART_ID,false,false);
    uart_set_format(UART_ID,DATA_BITS,STOP_BITS,PARITY);
    uart_set_fifo_enabled(UART_ID,false);
    // int UART_IRQ = UART_ID == uart0 ? UART0_IRQ : UART1_IRQ;
`

    // irq_set_exclusive_handler(UART_IRQ, on_uart_rx);
    // irq_set_enabled(UART_IRQ,true);

    // uart_set_irq_enables(UART_ID, true, false);

    
    while (true) {
        // rtc_get_datetime(&t);
        // datetime_to_str(datetime_str,sizeof(datetime_buf), &t);
        // printf("\r%s    ", datetime_str);
        // sleep_ms(100);
        gpio_put(LED_PIN, 1);
        sleep_ms(250);
        gpio_put(LED_PIN, 0);
        sleep_ms(250);

        while(uart_is_readable(UART_ID)) {
        uint8_t ch = uart_getc(UART_ID);
        printf("%s",ch);
    }
    
       
        // uart_putc_raw(UART_ID, 'A');
        // uart_putc(UART_ID,'B');
        // uart_puts(UART_ID, "Hello, UART!\n");
    }
}

// RX interrupt handler
// void on_uart_rx() {
    
    
 
    
// }`

Output: The garbage data
[image]

mcjurij, 2024-03-02:
Either baud settings wrong or baud rate too high.

Cheers,
Jurij

mcjurij, 2024-03-03:
See also: https://www.reddit.com/r/raspberrypipico/comments/uc9v1r/comment/i69n41m/

Cheers,
Jurij

engrzamanimran, 2024-03-04:
> Either baud settings wrong or baud rate too high. Cheers, Jurij
> […](#)

I have tried the communication with different baud rates and different board but issue remain the same.

peterharperuk, 2024-05-01:
Closing due lack of response to last comment

closed 2024-05-01 by peterharperuk