raspberrypi/pico-examples #480: Bug in BME280 SPI example when reading compensation parameters for humidity

ig1711, 2024-03-16:
According to the [BME280 Datasheet](https://www.bosch-sensortec.com/products/environmental-sensors/humidity-sensors-bme280/#documents) Section 4.2.2, the register addresses and compensation parameters are mapped as:
| Register Address | Register content |
|:-----------------------:|:----------------------:|
| `0xE5[7:4] / 0xE6` | `dig_H5 [3:0] / [11:4]` |
| `0xE7` | `dig_H6` |
 
But, in [bme280_spi.c](/raspberrypi/pico-examples/blob/master/spi/bme280_spi/bme280_spi.c#L173-L174), `dig_H5` is evaluated with the values of the registers `0xE6 and 0xE7` and `dig_H6` is evaluated with the values of `0xE8` (Note: `buffer[0]` is the value of `0xE1`, so, `buffer[5]`, `buffer[6]` and `buffer[7]` have values of `0xE6`, `0xE7` and `0xE8` respectively) i.e, not in accordance with the datasheet.

lurch, 2024-03-18:
Is this the same thing as #111 and #442 or are those slightly different?

peterharperuk, 2024-03-18:
The first is the dig_H1 problem I mention above, the second is different!

PR #481 (2024-05-01): Fix bme280 example
Fixes #480
commit fb3a85eb (2024-08-08): Fix bme280 example (#481)

Fix reading of dig_H5 and dig_H6
Fix reading of dig_H1
Fix order of reading of temp and pressure

Fixes #480
Fixes #111
Fixes #442