raspberrypi/pico-examples #489: host_cdc_msc_hid: incorrect len parameter in tuh_hid_report_received_cb

Franticware, 2024-04-13:
The len parameter is incorrect, it is 3 for many mice where it should be 4 or bigger. Essential data is then missing. For some mice, the len is correct, but that seems more like a coincidence.

Here is excerpt from hid_app.c with logging code to better see what is going on:
```

// Invoked when received report from device via interrupt endpoint
void tuh_hid_report_received_cb(uint8_t dev_addr, uint8_t instance, uint8_t const* report, uint16_t len)
{
  printf("len=%u;", (uint32_t)len);
  for (int i = 0; i != len; ++i)
  {
    printf(" %x", (uint32_t)report[i]);
  }
  printf("\r\n");
  uint8_t const itf_protocol = tuh_hid_interface_protocol(dev_addr, instance);

  switch (itf_protocol)
  {
    case HID_ITF_PROTOCOL_KEYBOARD:
      TU_LOG2("HID receive boot keyboard report\r\n");
      process_kbd_report( (hid_keyboard_report_t const*) report );
    break;

    case HID_ITF_PROTOCOL_MOUSE:
      TU_LOG2("HID receive boot mouse report\r\n");
      process_mouse_report( (hid_mouse_report_t const*) report );
    break;

    default:
      // Generic report requires matching ReportID and contents with previous parsed report info
      process_generic_report(dev_addr, instance, report, len);
    break;
  }

  // continue to request to receive report
  if ( !tuh_hid_receive_report(dev_addr, instance) )
  {
    printf("Error: cannot request to receive report\r\n");
  }
}
```

peterharperuk, 2024-05-01:
The length comes from tinyusb? I have seen a length > 3...

#0  process_mouse_report (report=0x20000fae <_hidh_dev+154>)
    at /home/peterh/source/pico/pico-examples/usb/host/host_cdc_msc_hid/hid_app.c:218
#1  0x100004fa in tuh_hid_report_received_cb (dev_addr=1 '\001', instance=1 '\001', report=0x20000fae <_hidh_dev+154> "", 
    len=4) at /home/peterh/source/pico/pico-examples/usb/host/host_cdc_msc_hid/hid_app.c:113
#2  0x1000c808 in hidh_xfer_cb (dev_addr=1 '\001', ep_addr=130 '\202', result=XFER_RESULT_SUCCESS, xferred_bytes=4)
    at /home/peterh/source/pico/pico-sdk/lib/tinyusb/src/class/hid/hid_host.c:299
#3  0x10009720 in tuh_task_ext (timeout_ms=4294967295, in_isr=false)
    at /home/peterh/source/pico/pico-sdk/lib/tinyusb/src/host/usbh.c:456
#4  0x1000098a in tuh_task () at /home/peterh/source/pico/pico-sdk/lib/tinyusb/src/host/usbh.h:118
#5  main () at /home/peterh/source/pico/pico-examples/usb/host/host_cdc_msc_hid/main.c:54

This code is just an example. Feel free to fix your code. If you have a suggestion for a specific improvement we could consider it.

Franticware, 2025-03-05:
This example still does not work (see my original post). What HID devices was it tried with and what was the output? I suggest trying to turn the mouse wheel for example to see if it the value of the axis is output (hint: it is not), my snippet will be useful for that. After we gather some actual data on this, this issue can be closed. It can then turn into issue for both tinyusb and pico-sdk, because it is a problem with compatibility between those two.

lurch, 2025-03-05:
pinging @hathach to see if he has any suggestions from the TinyUSB side.

Franticware, 2025-03-05:
I will add expected and actual data of my collection of USB mice in the following days. Some are rather esoteric, descriptor-wise (like 12-bit axis, or two-axis wheel). I managed to get quite a few working with older version of https://github.com/sekigon-gonnoc/Pico-PIO-USB , but the author switched from his own host stack to tinyusb and now it has the same issue.

Franticware, 2025-03-25:
First, I will provide the necessary patch to output report data.

Franticware, 2025-03-25:
This patch adds output of (potentially flawed) report data as provided by TinyUSB. I had to gzip it because it would not attach here in text format.
[print_report.tar.gz](https://github.com/user-attachments/files/19455804/print_report.tar.gz)

Franticware, 2025-03-25:
First mouse to analyze is Logitech RX300
Example output after patched is:
len=3; 0 ff 0

(-1 0 0)

Report data is 3 bytes: buttons, dx, dy
This is obviously wrong. The mouse has scroll wheel for which the data is missing completely, so that means the report is truncated.

This alone is reproduction of the issue. However, I am going to provide more info such as report descriptor and actual report data, how they should look:

Franticware, 2025-03-25:
Next one is this strange mouse:
 C-Tech Cronus Ultimate (GM-12) 

[image]

Franticware, 2025-03-25:
Logitech M-BT58
lsusb: Bus 001 Device 019: ID 046d:c03e Logitech, Inc. Premium Optical Wheel Mouse (M-BT58)
Similar to RX300, but it is even simpler (simple wheel, no tilt)
RPI output:
len=3; 0 0 1

(0 1 0)

usbhid-dump:
001:019:000:DESCRIPTOR         1742941718.359465
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 03
 15 00 25 01 95 03 75 01 81 02 95 05 81 03 05 01
 09 30 09 31 09 38 15 81 25 7F 75 08 95 03 81 06
 C0 C0

parsed:
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x03,        //     Usage Maximum (0x03)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x03,        //     Report Count (3)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x05,        //     Report Count (5)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x03,        //     Report Count (3)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection

// 50 bytes

Quite straightforward:

00fe0200
00fe0200
00fe0300
00fe0300
00ff0200
00ff0300
00000200
00000300
00010100
00000200
00010200
00010000
00000100
00010100
00010000
00010000
0000ff00
0000ff00
0000ff00
000000ff
0000ff00
0001ff00
0001ff00
0000ff00
00010000
0001ff00
0001ff00
01010000
0100ff00
0101ff00
0100ff00
0100ff00
0100ff00
0100ff00
00000000
00000100
00ff0000
00ff0000
00ff0100
00ff0000
00fe0200
00ff0000
00ff0000
00fe0100
00ff0100
00ff0000

4 bytes:
buttons, dx, dy, wheel

Still, one more byte compared to RPI example output.

Franticware, 2025-03-25:
Please let me know if adding data for more USB mouse models would help.

Franticware, 2025-04-21:
The key to solving problems with host_hid_to_device_cdc is using the following as part of TinyUSB initialization:
tuh_hid_set_default_protocol(HID_PROTOCOL_REPORT);

Then the report data must be interpreted according to the descriptor.

The default HID_PROTOCOL_BOOT does not work very well with many USB mice that I have tested so far.