raspberrypi/pico-examples #491: Bug in pio/uart_rx /uart_rx_intr.c

mxyxbb, 2024-04-16:
The function `irq_get_exclusive_handler ` return NULL if `pio_irq `can not be get.
And then `pio_irq++;` will not be run.

```
if (irq_get_exclusive_handler(pio_irq)) {
        pio_irq++;
        if (irq_get_exclusive_handler(pio_irq)) {
            panic("All IRQs are in use");
        }
    }
```
[https://github.com/raspberrypi/pico-examples/blob/eca13acf57916a0bd5961028314006983894fc84/pio/uart_rx/uart_rx_intr.c#L136C1-L142C1](https://github.com/raspberrypi/pico-examples/blob/eca13acf57916a0bd5961028314006983894fc84/pio/uart_rx/uart_rx_intr.c#L136C1-L142C1)

peterharperuk, 2024-04-16:
I think you're misinterpreting what the code is doing. It's checking if there is an irq without an exclusive handler. So if the first call to irq_get_exclusive_handler indicates that there's an exclusive handler, it checks the next irq. If they both return non-null then there are no free irqs. If irq_get_exclusive_handler return NULL then there's no handler and the code is free to use it. At the end of the code snippet pio_irq is the number of a free irq.

closed 2024-04-16 by peterharperuk

mxyxbb, 2024-04-16:
if I call function stdio_init_all() many times, it will fill up shared irq slots in rp2040 mcu.

If it is possible to offer a stdio_deinit_all funtion?

assertion "irq_hander_chain_free_slot_head >= 0" failed: file "H:\shared_files\pico\pico-sdk\src\rp2_common\hardware_irq\irq.c", line 215, function: irq_add_shared_handler

lurch, 2024-04-16:
> if I call function stdio_init_all() many times, it will fill up shared irq slots in rp2040 mcu.

Why are you calling that function multiple times? I believe the intention is that you'd normally call it just once, when your program first starts up.

Looking at the Pico SDK API documentation in general (e.g. [Hardware APIs](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html) or [High Level APIs](https://www.raspberrypi.com/documentation/pico-sdk/high_level.html) ), there appear to be many more `_init` functions than there are `_deinit` functions.

mxyxbb, 2024-04-17:
Since I want to printf some information before and after I overclock the rp2040, may I initialize the stdio again as the sys clock has been changed?
If I initialize it only once before I overclock the chip, uart baudrate will  go wrong after the overclock.
If I initialize it only once after I overclock the chip, printf info will not be visiable before the overclock.
If I initialize it twice before and after I overclock the chip, the init func will fill up shared irq slots in rp2040 mcu, which cause no shared irq can be used in my user function.

lurch, 2024-04-17:
In that case you should probably create a new issue in the `pico-sdk` repo, rather than continuing to discuss things on an already-closed issue in the `pico-examples` repo :wink:

peterharperuk, 2024-04-17:
https://github.com/raspberrypi/pico-sdk/issues/1688