raspberrypi/pico-examples #492: ir_nec variable length commands

omr-htp, 2024-04-28:
Regarding ir_nec example.
How to send a variable-length command.
My AC. for example, has 13 bytes command length. and it's NEC compatible (timing, leading mark and space...).

lurch, 2024-04-29:
Pinging @mjcross who was the original author of that example.

mjcross, 2024-04-29:
The NEC spec supported by the example (see [here](https://techdocs.altium.com/display/FPGA/NEC+Infrared+Transmission+Protocol)) is for a 67.5ms frame comprising:

- a 9ms sync burst,
- a 4.5ms space,
- four 8-bit bytes of data bits,
- a 625.2us end-of-message burst.

'Repeat' bursts are also possible but do not contain any data.

[image]

mjcross, 2024-04-29:
Please bear in mind that these are examples of how to use the PIO, not production code.

It's seems unlikely that your AC requires a 13 **byte** command length. If you want to investigate the codes sent by your remote, consider modifying your local copy of `pio/ir_nec/ir_loopback/ir_loopback.c` to comment out the transmit code and display the binary value of `rx_frame`.

mjcross, 2024-04-29:
@omr-htp my original repository [https://github.com/mjcross/rpi-pico-nec-ir-transmit-pio](https://github.com/mjcross/rpi-pico-nec-ir-transmit-pio) (older, and no longer actively maintained) has a little more information on how to use the PIO driver to transmit NEC extended and raw frames.

omr-htp, 2024-04-29:
@mjcross thanks for your reply.
I'm using LIRC at raspberry pi to get codes from IR remotes (.NET core app).

when using a standard NEC remote, I got 67 Pulse Space Durations or (Units) starting with 16, 8 then 1 or 3 as you know,
by using my AC. remote, I got 211 units contains same leading and bits (16,8,1,3.........) but longer.

by the way I did transmit commands using GPIO (I can share if it helps), but sometimes a timing issue appears. so, I have to use PIO which I didn't use at all.

the `tx_frame `at the old repo is 32 bits too as I understand.

Thanks.

omr-htp, 2024-04-29:
It's not an issue. I think we can tag it as a request

lurch, 2024-04-30:
I don't have commit-perms on this repo, but I think @peterharperuk does :slightly_smiling_face:

closed 2024-05-01 by peterharperuk