raspberrypi/pico-examples #503: What does example iperf.c

hasenradball, 2024-07-01:
@peterharperuk 
Dear Pi team,

sorry for the stupid Question what does the example 👍 picowiperf.c

It is as it is … no single line as a comment in the code!!!

lurch, 2024-07-01:
See https://github.com/raspberrypi/pico-examples/blob/master/README.md#pico-w-networking and https://www.google.com/search?q=iperf

hasenradball, 2024-07-01:
👍 thank you so much.

hasenradball, 2024-07-01:
@lurch
I tested the code but is there any detailed description how it works?
Do I need two Picos for running the test???

peterharperuk, 2024-07-01:
It's an iperf server. Just run an iperf client on another pc that can see it iperf -c <up address>
There is a way to build the code as a client. That's left as an exercise for the reader

hasenradball, 2024-07-01:
@peterharperuk 

thank you understood.
A little comment in code (kind of abstract) would helb here.
:-)

hasenradball, 2024-07-01:
@peterharperuk 

Just a short question.
Is it easy to build an PicoW code which can send an Http GET/POST Request every hour?

And is it possible to save data on the pico in a file.

For example read every 15 min. some sensor, but send it just every hour?

lurch, 2024-07-01:
@hasenradball I feel like you're asking questions which ought to be asked on the forums https://forums.raspberrypi.com/viewforum.php?f=143
GitHub issues is used for tracking bugs.

peterharperuk, 2024-07-01:
> Is it easy

Define easy

hasenradball, 2024-07-01:
By usage of already available functions for
e.g.: http requests
etc…

peterharperuk, 2024-07-01:
There's a http example here https://github.com/raspberrypi/pico-examples/pull/385

hasenradball, 2024-07-01:
I just looked into the official examples.
Thanks.

hasenradball, 2024-07-01:
One final question

in the wifi examples there are 2 executables

a bsakground version
and
a poll version

whats the difference?

closed 2024-07-02 by peterharperuk